fesch/CanZE #665: ISOTP 7-byte single frame

landswellsong, 2020-12-24:
Hey-hey,
My friend is making a custom CanZE build with a few extra command on his non-OEM ZOE BMS and we found out that apparently (?) there's no support for sending 7 byte single frame ISO TP. 

Code in question: https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/devices/ELM327.java#L596

I told him to try changing the clause to 14. Having worked with ELM low-level a bit, I know it might be related to the inability of sending a 8 byte CAN frame without `ATAL` command, but he tried and it worked so apparently it's executed somewhere down the line (I tried lazily grepping but didn't find).

So my question is, what's the logic behind that? Is it a compatibility thing to support dongles that can not into `ATAL` or is this something worth sending a pull request for?

yoh-there, 2020-12-24:
TBH, I think it is simply a long standing bug. I'll have a peek again re ATAL, but sending 7 bytes in a FIRST/NEXT sequence is pretty much against the specs. Man, this has been in there for AGES, probably since we added outgoing long messages (so that must have been TPMS support).

PR is welcome of course!

landswellsong, 2020-12-24:
Hmm what's the spec in question?

yoh-there, 2020-12-24:
Well it's wiki https://en.wikipedia.org/wiki/ISO_15765-2 But it says

> SINGLE: The single frame transferred contains the complete payload of up to 7 bytes (normal addressing) or 6 bytes (extended addressing)
> First frame: The first frame of a longer multi-frame message packet, used when more than 6/7 bytes of data segmented must be communicated. The first frame contains the length of the full packet, along with the initial data. 

We have never used extended addressing (used for gateway stuff; we might need to in the future), so read 7 above.

landswellsong, 2020-12-24:
Ah, I never worked with extended ids that's why I was confused. How about I implement this:
1. If the packet is extended addressed we keep as it is.
2. If the packet is standard addressed we allow 7 bytes too

Given `ATAL` is there, no other changes should be necessary. I'll build tomorrow and check if anything else falls of on a real car.

yoh-there, 2020-12-24:
Extended is a completely different beast than 29 bits addressing if I read correctly, so IMHO only the if @596 should be changed (12 -> 14)?

landswellsong, 2020-12-24:
Oh, that's another *extended*, yeah then the simple clause change with a comment would do.

yoh-there, 2020-12-24:
Thank you sir! No PR needed. Will put it in.

yoh-there, 2020-12-24:
Closed through 7214774c

closed 2020-12-24 by yoh-there