fesch/CanZE #666: Reset DTC missing a byte

landswellsong, 2020-12-24:
Based on feedback from the same friend ( @Dimonlipko ) who services Zoes, the reset DTC is set in the CSVs as:
```
14ffff
```
Whereas it should be
```
14ffffff
```

I personally can't confirm because I never reset DTCs, but he says it would be a welcome change since he is switching to BT terminal all the time from CanZE to do that.

Is this a bug or is the DTC reset as is working for your setups?

yoh-there, 2020-12-24:
He is right, it's a typo that got copy/pasted all over the place and actually we changed that in all the files a couple of weeks ago. I'll double check if we missed something. Are you on the most recent branch/commits?

yoh-there, 2020-12-24:
Never mind, missed a lot, though those are not used I think. Still sloppy. Thanks again!

yoh-there, 2020-12-24:
Fixed and pushed 911084c6

closed 2020-12-24 by yoh-there