fesch/CanZE #672: CanZE not working with brand new Konnwei

ledsales, 2021-01-03:
Just received a new Konnwei KW902 (from https://www.ebay.com.au/itm/133149170975), plugged it in to the Zoe ZE40 Intens (in Australia, a 2019 model so assume it is R90), and I get no usable data from the dongle. The BT and other LEDs flash (power LED is on solid) but just keep getting debug failed message at the bottom of the screen and various other error messages like incorrect data returned etc.

I have checked the resistance between pins 6 and 14 and it is 120 ohm, so it isn't the 20R resistor issue. I seem to be having the same issues as mentioned at https://github.com/fesch/CanZE/issues/655

Just wondered if anyone had any ideas on this, maybe there are dodgy versions of the KW902 circulating with problems other than the 20R resistor?

yoh-there, 2021-01-03:
If the connector for the dongle is in the center console under the 'coin box', it's indeed a '90 model, either Q or R but that doesn't matter for this.

It's close to impossible to do "remote analysis" on what's going on but here are a few things you can do, some of which are in the "rather silly" category, but what can we do eh?
- ensure the dongle is properly seated. Sometimes this is really hard and if one CANbus line is not connecting well, this could happen;
- ensure latest CanZE, 1.56 as I write this;
- flashing indicates it has been paired, and also properly selected in CanZE, so no issues there;
- run the ELM test and post a screen shot;
- switch on "debug" and let the app do something for a minutes, then close it and post the file. It should be in something like /android/data/lu.fisch.canze/files/debug......-log. 

We have not heard yet about 902's not working, but this sort of stuff is constantly being ripped and/or cost optimized. Anything is possible. A while back I had a 3 euro generic blue case one that DID work and had an actual PIC processor. The dongles are really a PIA :-( Anyway, happy to help what we can and do please keep us posted either way.

ledsales, 2021-01-03:
Thanks, did make sure the dongle was seated, also have the latest CanZE version, will run the ELM test and debug asap, maybe tomorrow, when I next go out to the garage I guess. Might also test the dongle in the subaru with Torque and see if it communicates correctly.

yoh-there, 2021-01-17:
Any news on this one?

ledsales, 2021-01-17:
Sorry, no, just haven't had time to think about it, about a hundred other things taking priority at the moment, and I'm back to work this week, so have no idea when I will get to it. Couldn't test it on the subaru as the suby had a flat battery from sitting in the driveway not being used (and the batt was 6 years old), only replaced that on friday...

yoh-there, 2021-01-17:
No problem. Closing this issue for now, but feel free to add or reopen it.

closed 2021-01-17 by yoh-there