fesch/CanZE #675: Zoe 50 Charge history Type column inconsistent

codavu, 2021-01-25:
Hi guys, 

I received my new Zoe 50, showroom car, bought with 750 KM on dash. 

Still I did one 22KW charge from 39% to 81% ( at 881 KM ), then tried the flexycharger at home and charged at 1.8 KW for 1.5 Hours up to 83 %  ( at 887 km / 85% on dash ). 

Still the problem I spoted is the column of Type, which seems somehow blocked to the values. 

Attached are the screenshots of the Charge History where the column can be seen to be stuck at the same values. 
I am using a Konnwei dongle, on Android 10

Thanks for the work done on CanZE !! Appreciated !

.

[image]
[image]
[image]

yoh-there, 2021-01-25:
Thanks for reporting!

CanZE tries to refresh these fields (60 in total) every 6 seconds. This might be too fast and the poller might fail to schedule in sequence. I will move the requests to the queue that handles them in sequence. I will close this issue, but please reopen if the issue persists after the current (1.59) release. 09ef6157

codavu, 2021-01-26:
Thanks for the quick fix on this one. I will keep an eye on this. 
Also I am new Zoe user, and if I can help on the quality part, I will do it.

supiiik, 2021-02-05:
@codavu can you please check, what FW version of EVC ECU you have? Because in my ZE50, the charging table isn't working at all

R240Pe, 2021-02-07:
I have been using CanZE every day for years for a complete battery documentation of all trips and charging processes for my Q90 and it is excellent that CanZE has also been adapted for ZE50 in recent months. Very good work! Congratulations!
Unfortunately, the charging history does not work on the ZE50. After completion of a loading process, the values ​​of the loading process are displayed. If you immediately carry out another loading process - without interrupting CanZE - this data is also displayed correctly and additionally after completion - i.e. 2 lines with loading data.
If CanZE is then terminated and restarted, there is no longer any access to the previously displayed data, all values are 0.
When the charging process is complete, the available range is also displayed. This range is not displayed on the charging info pages (NaN). The total battery range driven is also not displayed.
Is there a particular problem with showing these existing two data values ​​on both sides?
If you need further information or screenshots, I will be happy to provide it.
I have BETA and every release now 1.60 installed immediately.

yoh-there, 2021-02-07:
OK, so not fixed. Thanks for the additional info. Needs some discussion in the team. What we seem to know right now is:
- We have NOT been able to get the "GOM" at all from any ECU, which hampers the third graph in Consumption. So, I am quite curious you got it. Can you please give every detail you have? EVC firmware, what screen/field you got it?
- The history seems to be an EVC firmware problem. One ZE50 team member with fairly recent EVC firmware seems to have good values, another with somewhat older has not. I'll ask for their firmware numbers.

R240Pe, 2021-02-08:
EVC Data: 
diagVersion: 07
supplier: CONTINENTAL_ENGINE_SYSTEM
Soft: 0A10
Version: 6A00
I am attaching the screenshot with the loading progress of the two consecutive loading processes. The odometer reading 3096 displayed in the charging history was correct during the charging process. This number is not displayed on the "Technology load parameters" page, but the information "-".

Unfortunately, I didn't take an instant screenshot with the range at the time. So far, I only had a value displayed on the technology-charging-info-page this once, immediately after completing a charging process. If this information is not (so far) available for CanZE, the page may not have been completely updated and the line still showed a different or old value. Then I may have been wrong. But I will keep an eye on it during the next charging processes.

yoh-there, 2021-02-08:
Thank you 0A10 seems to indeed have this problem. Waiting for the results of the car that seems to be fine.

codavu, 2021-02-08:
Hey guys, 

I will give it a try today/tomorrow , once my wife gets home. It's her car.. so not direct access to it.

codavu, 2021-02-08:
Hi guys,

Managed to get my hands on the Zoe, and I have this:
[image]

As for the charge history I can still see the same "Inc / 3x Lente " repeating , as below:

Notice-Jeroen-Meijer, 2021-02-14:
Update, Remember, this applies only for the ZE50.

All drivers with EVC firmware version: 6A00, diagVersion: 07 seem to have this problem. Drivers with firmware version: 69B0, diagVersion: 08 have no problem with history. Note that the firmware version is LOWER, but the diagVersion is HIGHER.

I don't think there is anything we can do in CanZE to fix this. Unless someone sees another way out, or sees this problem on said firmware, this issues is earmarked for closing. I will change the title.

yoh-there, 2021-02-14:
Update, Remember, this applies only for the ZE50.

All drivers with EVC firmware version: 6A00, diagVersion: 07 seem to have this problem. Drivers with firmware version: 69B0, diagVersion: 08 have no problem with history. Note that the firmware version is LOWER, but the diagVersion is HIGHER.

I don't think there is anything we can do in CanZE to fix this. Unless someone sees another way out, or sees this problem on said firmware, this issues is earmarked for closing. I will change the title.

codavu, 2021-02-15:
Hi @yoh-there 

This issue was opened for the inconsistent Type column values. 
With each recharge I am doing I do get the same pattern of: Unknown / Slow / Slow / Slow  and repeating no matter the kilometers.

Is this inconsistency coming from the EVC ? I have 69B0 as firmware version. Or this is because of the dongle being used ?

Regards,

yoh-there, 2021-02-15:
@codavu I honestly can't answer that question. I need to relay it to ZE50 owners as I cannot check it at all.

To the team: sorry guys ;-)

codavu, 2021-02-15:
@yoh-there No problem 

Thanks for the great work you are doing on CanZE !!

R240Pe, 2021-02-19:
As I wrote some days ago: "The total battery range driven is also not displayed." on the CanZE Screen "Ladeparameter - HV-Akkudaten - gefahrene Akkureichweite -"
I knew, I saw it somewhere - and now I found this information: **EVC-HCM-VCM 18daf1da.622006.24, Total vehicle distance,3253,km**

[image]

yoh-there, 2021-08-29:
This issue has gone stale. I am closing it but feel free to reopen. Personally, there is not a lot I can do with my Q210.

closed 2021-08-29 by yoh-there