fesch/CanZE #676: Significato di Total battery usage

casalelli, 2021-01-25:
Il total battery usage è sono i kWh caricati nella batteria compresi dell'energia rigenerata o no?
Per il consumo totale dell'auto devo considerare 
Total battery usage
oppure
Total battery usage - total energy from regenerative breacking
?
Nota: breacking dovrebbe essere braking
grezie e grazie del vostro lavoro

yoh-there, 2021-01-25:
We cannot be 100% sure, but since this number is reported by the battery computer, I don't think it has knowledge that it's coming from regeneration or not. Hope this helps.

closed 2021-01-25 by yoh-there

casalelli, 2021-01-25:
Sono riuscito a fare 2 prove, con l'utilizzo aumenta la sola energia da rigenerazione di frenata per cui la Total Battery usage è la sola energia dovuta alle ricariche esterne