fesch/CanZE #677: Can CanZe be adapted to Captur plugin adding a specific profile?

Roberto-1967, 2021-01-27:
Hi I know that this is not an issue but CanZe works at 99% on my captur plugin using ZOE50 profile but there are issues in some parameters so the overall experience is not good. It's really a pity since most of the features are perfectly running.
Do you think it's possible to add a "Captur plugin" profile?
I can help you somehow providing logs from my car if needed.
Some screenshots below.
[image]

 Have a look to the Nm
[image]

[image]

This is working as most of the other features
[image]

fesch, 2021-01-27:
Is there a Captur EV version?

Roberto-1967, 2021-01-27:
it's a plugin 9.8 kw battery

fmarsico70, 2021-01-28:
I add myself to Roberto's request for this improvement.

yoh-there, 2021-04-16:
While I honestly understand and appreciate the interest, I don't think any of the current developers are able to pick this up, manly by simply not having access to the car. We learned the hard way that it is close to impossible to get things right without that. I propose to close the issue, as it's basically stale.

closed 2021-05-06 by yoh-there

Roberto-1967, 2021-05-07:
Ok I understand the difficulties related to not having the car to run test. 
Let me say that if any existing or new developer will have an E-tech car available it could be very interesting to extend the coverage in a dedicated branch. 
You'll notice that there will be a lot of e-tech cars on the roads in the next months.
Thanks