fesch/CanZE #679: ZE50 StartDiagnosticSession.Extended

supiiik, 2021-02-04:
Hi, I'd like to know, why are you starting every session with StartDiagnosticSession.Extended 0x1003 command? I found, that it's working even without that. I'm programming my own device, which is sending a few values from LBC to my EVSE and it's working even without Extended session. Beside that, if I open Extended Diagnostic session and the car is switched off with that session opened, then at the next vehicle start are weird messages on the display. I think, that you are using same manner in CanZE, because when I leave ODB connected while car is switched off, that at the next start it's displaying also same weird messages and allways reset my trip meter.
Maybe you should properly close Extended session with 0x1001 command or if possible (I don't know), don't even use it. I'm not android programmer, so I can't test it with CanZE, but as I said, I'm reading 10 commands from LBC and they are working even without 0x1003 start command
(sorry for my bad english)

yoh-there, 2021-02-04:
Hi,

I assume you are talking about sending that command to the S-GW ECU. We deemed in necessary to open the gateway. However, there is some debate if it's really needed. More tests required.

The BCB most certainly needs the command.

The other ECUs are not send that command AFAIK. If you think otherwise, please let me know.

Since the older models do not have a S-GW, the only ECU that receives this command is the BCB.

I'll leave this issue open as some more tests re. the S-GW. As I do not own a Ph2 car myself I can't comment on the timeline.

supiiik, 2021-02-04:
Hi,
Yes, I'm talking about S-GW ECU where you sending 1003 ID. I have tested only LBC and EVC ECUs and these don't need extended session opened. Don't know about the others. I was sniffing PyClip on CanBus and they for example sending 1003 to LBC ECU (when I'm reading from LBC), not to S-GW ECU as CanZE. 
But the problem is, that it's causing trip meter resets and also weird warning messages on the car display. It was doing also my device, but today I tryed to not use it with extended session opened and the problem seems gone. I will try to test it more in next days.
Unfortunatelly, I'm not android programmer, so I can't compile updated version without DiagnosticSession to test, what ECUs working and what are not in CanZE :(

yoh-there, 2021-02-04:
No worries, we'll do some investigations and make changes if needed for sure. BTW I am sure the LBC doesn't need it, it's very forgiving.

yoh-there, 2021-08-29:
We'll probably stop sending commands to the gateway in the next version. Tests seem to indicate it is not necessary after all. This change is already in the development branch. It will be released in 1.63. Closing. Thanks for your contribution.

closed 2021-08-29 by yoh-there