fesch/CanZE #681: Zoe R110 PH2 does not work

Starter4711, 2021-02-16:
Hi, the Amazon Dongle (decribed in the german video) ist not compatible with iOS. On Android I do not get one Information out of the car.
Regards.
https://www.amazon.de/gp/product/B079N9KD4P/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&psc=1

ik2wxx, 2021-02-16:
Hello
Please do not confuse our app with the another app with same name that's not from the CanZE Team.
Our app is very new and we didn't do any video about it, so I think the video is about the other app.
Supported OBD for our app are:
Konnwei wifi and similar
LeLink BLE
VGate iCar Pro BLE
(should work with CanSee on Wifi)

ik2wxx, 2021-02-16:
BTW Android app is not compatible with any BLE dongle

Starter4711, 2021-02-16:
Hmm, OK. But do not understand, your link in the readme is to the App I am using?

ik2wxx, 2021-02-16:
Our app is available in beta test from yesterday.
The older app (I suppose you're referencing to that) is a sister project from a developer not related to the original team. 
We have references to the older app on the site, but we will delete them as the official app will be ready.
We can't help you with the older app, please contact the developer.
If you want to try the official app, the link is here https://canze.fisch.lu/canze-for-ios/

closed 2021-02-16 by ik2wxx