fesch/CanZE #683: ZE50 error while reading LBC DTCs

ZMan88, 2021-02-18:
Hello,

I have an issue with CanZe when trying to read the DTCs codes for the LBC
[image]
Not sure if it is a problem with the zoe, or the dongle used.
The dongle used is KONNWEI KW902.

I also have errors when trying to get all the firmware versions 
[image]

Also getting errors when opening the BRAKING activity but did not get a screenshot of that.
If there is any more info that I could provide, tell me and I will.

Love the app btw, adds a lot of functionality that Renault missed when developing the system.

ZMan88, 2021-02-18:
I forgot to say, I m using version 1.60

yoh-there, 2021-02-19:
What dongle are you using please? Something is very wrong if it doesn't accept atsp6.

ZMan88, 2021-02-20:
The dongle i am using is
[image]

ZMan88, 2021-03-03:
[image]
I get this when I test the dongle, if it helps

Leobaillard, 2021-03-04:
Thanks for the details. I have some tests planned in my Zoe with the Konwei. I'll check your issue and report back as soon as possible.

yoh-there, 2021-08-29:
There has been no follow up on this issue. I will close it because it is stale. Feel free to reopen.

closed 2021-08-29 by yoh-there