fesch/CanZE #684: ZE50 errors with canze v 1.60

AsparagusRex, 2021-02-18:
Hello,

I am not a technical person but I try my best.
I connected an ELM adapter (KONNWEI KW902 OBDII) with my ZOE Phase 2, Life R110 Z.E. 50.
I eanbled bluetooth on the adapter and my smartphone and got agood connection.
Selected the Konnwei adapter and the ZE40/ZE50 in the canze app.
So far so good.
Bit on most screens i get error messages and no data :(

I ataach two screenshots.

Any idea, what is wrong here?

Thanks

As.
[image]

[image]

yoh-there, 2021-02-19:
Not all fields we know from the Ph1 cars are mapped or even available in the Ph2. But the does not exist should not happen. Can you add a screen of your setting, especially so that it shows your device and the car settings?

AsparagusRex, 2021-02-20:
is that what yu are looking for?

[image]

AsparagusRex, 2021-02-26:
Hello? Any ideas out there?

ZMan88, 2021-02-27:
I have the same errors on my zoe ze50 with konnwei adaptor

AsparagusRex, 2021-03-03:
Is it a problem of the adapter (Kinnwei), or the software, or the Zoe model?

Leobaillard, 2021-03-04:
Hi, I'm currently doing tests on my Zoe and a Konwei adapter to try and reproduce. I'll keep you updated.

AsparagusRex, 2021-03-06:
Let me know if I canbe of help for testing etc.

AsparagusRex, 2021-04-12:
Any progress on this issue?

Leobaillard, 2021-04-12:
Hi!

Sorry for not answering sooner, I had to take a break for family reasons and I'm now back.

I just did some more tests in my car with two different phones and the stable version of the app, everything is working as expected with the Konnwei.

Some follow up questions though:
- do you get any data while driving for the consumption screen for example?
- did you check the option in settings to allow data while driving?
- some of the data is only "moving" when the car is moving too, can that explain the data showing "0" on your end? (sorry if this seems trivial, but just making sure)
- could you show us the results of the "ELM test" page of the app while the car is ON and the Konnwei is connected?

Some more questions about the Konnwei:
- when did you buy the dongle?
- if you bought it online, do you still have the listing page with the product details and could you link it here?

I'm sure we can figure this out if there is indeed a problem related to the app. It could also be a newer version of the dongle that changed stuff in the way it communicates with the app (hence my questions about your dongle).

AsparagusRex, 2021-04-12:
Hello,

glad you are back.

Let me start with the Konnwei question :)

I bought it online at the Konnwei Store at Amazon:
https://www.amazon.de/gp/product/B077XGSV83

Now for the other questions:
>do you get any data while driving for the consumption screen for example?
Yes, screenshot attached
 
>did you check the option in settings to allow data while driving?
Yes

>could you show us the results of the "ELM test" page of the app while the car is ON and the Konnwei is connected?
attached

Hope this helps :)
[image]
[image]
[image]

Leobaillard, 2021-04-12:
Ok, good, thanks!

Just to be sure, what is not working? It seems from your screenshot that the data is loading, right? Do you only get the error message when opening the page first and not after or does it come back repeatedly while using a particular screen? For the former, it could just be that the data takes time to load initially for some reason but it ok afterwards.

AsparagusRex, 2021-04-13:
Let me test that and get back to you

AsparagusRex, 2021-05-20:
Hi, sorry it took me so long but now I am happy to report that everything seems to be working on the Z50.
Thanks a lot
As.

yoh-there, 2021-05-24:
Thank you for reporting back to us. Closing.

closed 2021-05-24 by yoh-there