fesch/CanZE #687: Which parameters are read from the car, and which are calculated?

Sebes1, 2021-03-05:
I complained in the service that the charging is finishes without reaching 100% SOC. In the dashboard I got 100%, but after couple of minutes it counting down. I could verify with official Renault app (My Renault) that the charge was finished without reaching 100% (usually 89-97%). And of course I could read out with your app, CanZE. But there are a difference 1-4% depending on the SOC level. I tried with ddt4all, and verified that CanZE SOC signing is OK.

I got (paid) the latest firmware update(s) of R240 ZOE. 
LBC 0854 (old 0852)
LBC2 0854 (old 0853)
BCB 0210/08C2 (old 0210/0871)
PEB 0220/7420 (old 0220/6500)
EVC 0204/0AC0 (old 0204/07C0)
I realised that the SOH level is continously changing, from 98% to 95%, than back to 97%, and so on. SOH levels are verified with CanZE and ddt4all. Couple of days before I got 93% from 97%.

When I had the old firmwares and used CanZE the followings were through:
100% SOC and 98% Real state of charge: Available Energy was 22.8kWh, and Energy to full was 0kWh.
I assume we know that total energy is 96x3.75V*72Ah=25.92kWh, and after first big BMS update the usable energy is 0.9x25.92kWh=23.328kWh. So, 100% SOC and 100% SOH means 23.3kWh. (in my case 0.98x23.3kWh=22.83kWh, OK)

But now, my Zoe has the following data:
SOH 93%
at 100% SOH Available Energy is 21.6kWh, and Energy to full was 0kWh. (should be 0.93x23.3=21.67kWh, which is quite OK)
However, Real state of charge is 98% (depending on the charging it varies between 96,5-98.0%). I misunderstand Real state of charge meaning?

Sorry for the long comment, but I would like to know what is calculated by CanZE, and what is read from the car?
Real state of charge?
Available Energy?
Energy to full?

I could not found these parameters in ddt4all. Please help me to find out what the problem could be. Thanks.

yoh-there, 2021-03-05:
- Except Energy to Full, all values are taken from the car and NOT calculated by CanZE.
- We have no semantic definitions of the fields, but the naming usually is enough to make a good enough guess.
- It is our unconfirmed understanding that the real soc is what you'd normally see in a battery management system, being the relative battery capacity to the (calculated) total capacity. Note that the total capacity is degrading. The user soc is the soc that is presented to the driver and represents what can actually be used, since for longevity and warranty reasons, the BMS keeps a bit of spare capacity at both the top and the bottom of the capacity.

Hope this helps somewhat.

Sebes1, 2021-03-05:
Thank you for the answer.

I try to understand what I see from the data.
I assume you have the Renault Trainee Booklet for the first gen ZOE.
https://www.google.com/search?q=renault+trainee+booklet&rlz=1C1FLDB_enHU563HU573&oq=renault+trainee+booklet&aqs=chrome..69i57j0i22i30.12064j0j7&sourceid=chrome&ie=UTF-8

We all know that first gen ZOE battery contains 12 modules, each modul consists 16 cells, and one cell is 3.75V/36Ah. I mentioned 96pcs of 72Ah cells, while two cells are paralel configuration. So the nominal energy in the battery is 192pcsx3.75Vx36Ah=96pcsx3.75Vx72Ah=25920Wh=25.92kWh. Attached an original battery label for verification:
[image]
At first, only 22kWh was available, 85% of the nominal capacity: 0.85x25.92kWh=22.032kWh (22kWh).
After first big BMS Update it was increased to 90% regardless of Q or R motor type: 0.9x25.92kWh=23.328kWh (23.3kWh).

The spar what you mentioned (15% or 10%) in the reality is a charging voltage limit (around 4.13V instead of 4.2V or more), and a discharging voltage limit (I have never tested, but I will).

I fully understand that battery degrades with time, and usage (charging and discharging cycles). And degradation depends on many other parameters (f.e. temperature, storage at different SOC, etc.).

In case of SOC, batteries loose capacity, not voltage, so 100% SOC means nothing about capacity, just that reached end of charge voltage. It could be a perfect battery or degraded battery.

But Real state of charge at 100% SOH means (or have to) the actual Energy compared to Nominal Energy. If you think that you read this value than my battery is still around 96-98%.

In my case 4% degardation (almost 1kWh) within one day is more like a software bug than a reality. I checked every cell voltages at different SOH level, and no single cell degradation can be observed. That is why I would like to know what data can I trust in CanZE.

FYI there are 2 different battery models in the ZOE regarding to ddt4all:
RSA simple model (98% in my car)
LG model (93% in my car)

And I can edit Ah capacity filed, and LG SOH field with this program. Maybe I will try it....

Sebes1, 2021-03-06:
It seems that the above mentioned firmwares are wrong.
I started to charge my ZOE with 93% SOH yesterday (-11.9kWh and 9kWh, total 20.9kWh), and it was fully charged in the morning with 95% SOH (21.9kWh). 2% (1kWh) healing in one night is technically impossible.
I think your app is OK, it shows the right values. Same for Renault again...
You can close this issue. However, if you have some comments, please let me know.

closed 2021-04-01 by yoh-there