fesch/CanZE #691: Please take contact with me

don-alberto, 2021-03-22:
I represent a company intending to invest in an OBD II solution and we would be happy to access your expertise on ZOE and OBD. Since I found no other ways to contact, please mail us under support_AT_teslametric.com
Please contact us as soon as possible, many thanks in advance.

fesch, 2021-03-22:
Hi Alberto

Everything knowledge we acquired is in our code, which you can freely look at, so I don't really get the point about your request.

don-alberto, 2021-03-22:
Thanks for getting back!
Yes sure, but we want to develop our own hardware for a bit different usage scenario. We however are lacking OBD side of know-how.

FulminatingGhost, 2021-03-22:
For hardware you might want to have a look over at the CanSee project here https://gitlab.com/jeroenmeijer/cansee

don-alberto, 2021-03-23:
That’s exciting, but we would need some modifications. E.g. we need a bigger ring buffer and some business logic on the board. We would like to hire somebody, who has deeper understanding for the problems

closed 2021-03-23 by yoh-there