fesch/CanZE #692: iOS TestFlight wife dongle connection settings

JulesBuh, 2021-03-30:
I have the iOS CanZE app build 3.0.1.0 which works with v-link WiFi dongle. I was trying to connect the TestFlight iOS app 1.0(5) but couldn’t match the settings to connect.

I’ve tried http option and match the up and port that the working app shows in the settings but that didn’t work on TestFlight iOS app.

the TestFlight one looks better since I’m trying to get the log which isn’t accessible on the 3.0.1 iOS build

any suggestions?

ik2wxx, 2021-03-30:
Hi
I can't talk about previous iOS version, I don't know how it works.

About TestFlight version, if the dongle is wifi, you should select "wifi" in the app settings, not http.
But first you should connect to the dongle wifi network.

So, first power up the dongle, wait for his wifi network, connect the phone to it.
Get the ip address of the dongle.
Configure the app to connect to a ELM327 dongle via wifi, insert dongle ip address and port, and test connection with the button in the settings screen.
Please let me know.

JulesBuh, 2021-03-30:
Thanks I’ll give that a go and report bsck

StevenMeel, 2021-05-06:
Bonjour 
j'essaye de me connecter mais sur les paramètres wifi  j'ai bien l'adresse ip mais je n'ai pas de port. 
[image]

ik2wxx, 2021-05-06:
Hello
Try 35000, that's the default tcp port for most wifi obd.
Otherwise, check on the adapter documentation.

StevenMeel, 2021-05-06:
Je viens de vérifier la notice du dongle et le port est bien 35000. 
Je dois faire quelques chose de mal, impossible de ce connecter. 
[image]
[image]
[image]
[image]
[image]

ik2wxx, 2021-05-06:
Hello
The adapter ip is 192.168.0.10, not 192.168.0.11 (this is your phone address)

StevenMeel, 2021-05-06:
J'ai corrigé mais je me connecte toujours pas. 
[image]

Voila ce que j'obtiens quand j'appuie sur connect puis test. 
Faut il que j'appuie sur d'autres boutons pour me connecter ?
[image]
[image]
[image]

ik2wxx, 2021-05-06:
Hello
The adapter is connected, and CanZE is correctly connected to it, but the adapter is not replying correctly.
Look at the "NO DATA" replies...
Is the car powered ?
Is it a Q210 model Zoe ?
Can you post a photo of the adapter ?
Thank you

StevenMeel, 2021-05-06:
J'ai compris mon erreur j'ai pas appuyer sur la prise. 
maintenant tout fonctionne merci
[image]
[image]

ik2wxx, 2021-05-06:
Hello
Please ask if you need more help
Bye

closed 2021-05-06 by ik2wxx