fesch/CanZE #693: iOS version CanZe 1.0(5). No data

mago-kamar, 2021-04-10:
OBD right connected, but no data retrying.
iOS version 14.4 iPhone X
CanZe version 1.0(5)
[CanZE 1618070992.log](https://github.com/fesch/CanZE/files/6290280/CanZE.1618070992.log)

Attched log files.

Please, let me know if I can supply other info,

Thanks

Enrico

ik2wxx, 2021-04-12:
Ciao Enrico.
Da come puoi leggere dal log, l'adattatore risponde "NO DATA" alla richiesta di lettura dei dati.
Questo potrebbe essere un problema dovuto all'auto non accesa (alcuni dati possono essere letti solo a motore acceso), oppure un problema dell'adattatore stesso.
Ti consiglio di riprovare a motore acceso, oppure a cambiare adattatore (purtroppo non hai scritto quale adattatore usi) usando uno di quelli consigliati (Konnwei wifi, Vgate icar pro ble e Lelink ble).
Grazie

Hi Enrico.
As you can read from the log, the adapter replies "NO DATA" to the data request.
This could be a problem with the car not powered up (some data can only be read with the engine running), or a problem with the adapter itself.
I advise you to try again with the engine running, or to change adapter (unfortunately you have not written which adapter you use) using one of the recommended ones (Konnwei wifi, Vgate icar pro ble and Lelink ble).
Thank you

mago-kamar, 2021-04-12:
Ciao Roberto e grazie per la risposta,
come adattatore, uso quello che utilizzavo sulla Zoe R240, un VGate iCar 2 WiFi.
Ho provato anche con auto in moto, ma il risultato non cambia.

Credi sia colpa dell’adattatore?

Grazie,

Enrico

ik2wxx, 2021-04-13:
Ciao Enrico
Come ultima prova ti consiglierei di provare con un altro adattatore (magari in prestito).
Purtroppo i comandi per la ZE50 sono diversi da quelli delle serie precedenti e non è detto che un adattatore che funziona bene con la R240 vada bene per la ZE50.
Facci sapere

Please try another obd dongle.

olieske, 2021-04-24:
I have the same issue. 
ZOE ZE40 (ph2)
Vgate iCar Pro BLE4.0
iOS 14.4.2 on iPhone 12
CanZE Version 1.0(5)

Setup says adapter connected, _Test dongle says "dongle K.O. :("

ik2wxx, 2021-04-26:
> I have the same issue.
> ZOE ZE40 (ph2)
> Vgate iCar Pro BLE4.0
> iOS 14.4.2 on iPhone 12
> CanZE Version 1.0(5)
> 
> Setup says adapter connected, _Test dongle says "dongle K.O. :("

Please check the log, that should tell you what's wrong.
Post the log if you need assistance.
(you can find logs in the "File" App, you will need to enable logs in CanZE Settings screen)
Thank you.

yoh-there, 2021-05-06:
Issue is stale. Closing.

closed 2021-05-06 by yoh-there