fesch/CanZE #694: CanZe logs

cad0p, 2021-04-10:
Hi, where are the logs stored?

I'm on Android. I found an earlier post but it didn't solve my issue. 

Thanks!

ik2wxx, 2021-04-12:
Hi
On my Android phone the logs are in the 
/storage/self/primary/Android/data/lu.fisch.canze/files
folder.
My Android OS version is 11.

cad0p, 2021-04-12:
No items there unfortunately

ik2wxx, 2021-04-12:
Please tell us you app version, OS version, device brand and model.
Thank you

cad0p, 2021-04-12:
Pixel 2 XL, Android 11, app version 1.60

ik2wxx, 2021-04-12:
Ciao
Please post a screenshot of app settings (with "LOGGING" section visible) and app authorizations screenshot (tap and hold app icon, select "app info/Informazioni App" then "Authorizations/Autorizzazioni").
I have Pixel 3 XL with Android 11 so I'm expecting they work in a very similar way.

yoh-there, 2021-05-06:
Issue is stale, closing.

cad0p, 2021-05-06:
Sorry, i thought I sent it!

cad0p, 2021-05-06:
[image]
[image]

Best-car, 2021-08-07:
Hi !
I have used CanZe also, but i see the same problem........i don't find the loggings !
The app makes the file......it says Wait, writing /storage/emulated/0/Android/data/lu.fisch.canze/files/Firmwares-2021-08-7-11-55-51.csv  and a few seconds later Done  !
But i cannot find the files on my device (Samsung tablet S5e) not in my SD-kaart if i choose this option in the settings but also not in the "interne opslag" if i don't choose the option "Log to sdcard1"  I did also try my Samsung smartphone S10 plus, but the same here. Both devices are using android 11.  The android/data map from my SD-kaart is and stays empty !

Hm....after typing the above text i thought, maybe i have to make the Android map.....but my device (S10+) says...."mapnaam al in gebruik" mapname already in use !!!!
Short google.......it seems that android 11 hids the Android map......if you are using the "mijn bestanden" (my files) app which is standard on my device......
I did find the files if i connected my S10+/S5e tablet  to my windows 10 computer with a USB cable.
Also i tried Total commander, also a given tip when i did the google search.....and that app also shows the Android/data folder.

So i am happy....found it !

Good Luck.

Best-car, 2021-08-07:
Still one thing..........

One problem to solve....the app doesn't write to the SD card, even not if you choose this setting option (Log to sdcard1)
The android/data map is on my sdcard.....but that stays empty...

Greetings Best-car

yoh-there, 2021-08-09:
Thanks for figuring that out. We need to do some clean up in the logging department it seems ;-)

Android 11 hiding that folder is a big PIA. They changed the entire storage architecture and we need to have a look at that again. Keeping this issue open.

@fesch could you have a think about the actual storage in relation to the Android upgrades? The mucking around with file paths is getting annoying ;-)

Best-car, 2021-08-09:
Thanks for the comment.

Indeed android 11 and hiding things is a big PIA.....hmm yeh, and i had changed the setting to "unhide system files"
but clearly that's not the hidden android map....a big PIA.

Let people self decide which path and which map name the would use and make, and this for internal ore external storage usage 
.......extra new settings options.

Greetz Best-car

fesch, 2021-08-20:
I've updated the code in the dev-branche as follows (for now):
- remove the "log to sdcard1" option
- all data goes to the internal storage, subfolder CanZE (same level as DCIM, Download, ...)

yoh-there, 2021-08-20:
Cool, thanks. However, I do read this

https://developer.android.com/training/data-storage/use-cases#opt-out-in-production-app

as being a sunset thing (opt in for Android 10, impossible for Android 11). Maybe I read it wrong.

fesch, 2021-08-21:
I definitely needs some Android 11 device ;)

Bu no, you are right. The actual code only works with Android 10.
The fact that Android 11 is more restrictive still remains a problem to be solved ...

yoh-there, 2023-10-02:
Closing (stale)

closed 2023-10-02 by yoh-there