fesch/CanZE #695: Inclusion in F-Droid

frh, 2021-04-30:
Would you mind if CanZE were included in the free and open source app store F-Droid (https://f-droid.org/)? This would be very helpful for those not running Google services.

For this a build flavor not using Firebase would have to be created, as F-Droid does not allow proprietary dependencies.

yoh-there, 2021-04-30:
I had a quick look at the process. It seems elaborate, adding the project to their git repository and F-Droid compiling from source. Quite another workflow than the Play store.

Firebase is a tool for us to do quality monitoring that we really like ;-)

Given the limited gains this doesn't seems to be a diligent use of our limited resources.

I have no problem to either
- restart our habit of publishing binary (.apk) releases on GitHub. (To be honest at some moment in time we simply forgot doing that and nobody seems to have noticed)
- informing you when a new release is out in case you want to take on this task yourself, and publishing a link to the F-Droid page.

yoh-there, 2021-05-01:
No problem at all. Lovely if you would take that on thanks. If we need to incorporate something for a flavor, feel free to do a pull request.

Don't hesitate to remind us if we seem to have skipped a release tag. We're on a fairly slow release schedule now, so it might fall through the cracks but from my end I'll try to put/keep it in the workflow.

frh, 2021-05-02:
Thanks for the merge! Regarding the release, it looks like the latest version is 1.60, but the latest tagged release is 1.48?

yoh-there, 2021-05-03:
Correct. As I said, we kind of forgot to do that. Does it help if I add a a 1.60 tag?

frh, 2021-05-03:
Adding a 1.60 tag would help a lot, thank you! Especially helpful if that tag includes the Gradle build flavor commit!

From what I understand F-Droid packaging only cares about the Git tag, so I don't think it matters whether or not the source code is included as a binary.

yoh-there, 2021-05-03:
Done :-)

frh, 2021-05-03:
Thanks! I just opened a Request For Packaging at https://gitlab.com/fdroid/rfp/-/issues/1724. I'll let you know if any issues arise.

yoh-there, 2021-05-06:
Done :-)

frh, 2021-05-06:
Thanks for the review! I will get started on a PR to F-Droid to package CanZE. Would you mind doing a new release tag to include these changes, or maybe removing and re-adding the 1.60 tag if that's possible? Sorry for that, my bad!

frh, 2021-05-06:
From the reply I got in the RFP ticket, I'm pretty sure that you can do it now.

yoh-there, 2021-05-07:
Done :-)

Notice-Jeroen-Meijer, 2021-05-13:
I don't think it's included yet. Anything I need to do?

frh, 2021-05-13:
> I don't think it's included yet. Anything I need to do?

Thanks, but it looks like all we're waiting for now is for the F-Droid people to review my PR at https://gitlab.com/fdroid/fdroiddata/-/merge_requests/8935.

yoh-there, 2021-05-24:
Looks like there is stagnation on the Fdroid end after your hard work. Indeed there seems nothing on our ends that they seem to be waiting for.

frh, 2021-05-24:
I asked about it on IRC. The reviewers are just busy, so we'll just have to wait a bit longer.

frh, 2021-06-07:
The PR is being reviewed now. They want to include info on what the app's permissions are used for. Based on my own digging these are what each permission is used for, is this correct?

**location (`ACCESS_COARSE_LOCATION`, `ACCESS_FINE_LOCATION`):** Include location in optional local logging
**Bluetooth (`BLUETOOTH`, `BLUETOOTH_ADMIN`):** Car communication
**storage (`WRITE_EXTERNAL_STORAGE` and thus implicitly also `READ_EXTERNAL_STORAGE`):** Optional writing of logs
**network (`INTERNET`):** Get app news from GitHub repo

yoh-there, 2021-06-07:
That is all correct. I would probably say "Canbus dongle communication", but that's nitpicking. Have they indicated anything about where to put that information? Let me know if I need to pull something please.

yoh-there, 2021-06-07:
Merged. Do they need a version bump?

frh, 2021-06-07:
Thanks! Unfortunately they do need a bump. I'll ask if there's anything else they need so we can avoid any more unnecessary ones.

frh, 2021-06-08:
No more changes here should be necessary, so if you could do a version bump to include the latest PR that would be great.

yoh-there, 2021-06-08:
Done :-)

frh, 2021-06-08:
Thanks! Could you tag the new release as well?

yoh-there, 2021-06-08:
Sorry. Done.

yoh-there, 2023-10-02:
I think F droid use is stale. On our end we forgot tagging releases, on the other hand, no one raised an issue. I am closing this one for it going stale.

closed 2023-10-02 by yoh-there

frh, 2023-10-11:
> I think F droid use is stale. On our end we forgot tagging releases, on the other hand, no one raised an issue. I am closing this one for it going stale.

The F-Droid build is still working very well. I've been using it for a few years now, but I never went to GitHub to check for new releases so I didn't notice that the version was stale. It's set to auto release new versions when tagged on GH with the version number as the tag name, so if you tag a new release it should get picked up!