fesch/CanZE #697: CanZE 1.60 (android):  query IDs of already paired TPMS valves on new ZOE ZE 50

Zoe-Fahrer, 2021-05-03:
At first: great compliment to the very useful CanZE App! 
I used it last four years with my ZOE R240 and it enabled me to change summer / winter tires. 

But since I changeg car to the new ZOE ZE 50 it does not longer work.
Can anyone advice me please, how to read the IDs of my winter tires?
With android Version 1.60 (release-master) Build: 2021-02-04 all I see, is the screen "Information zu den Rädern" without any value.
CanZE can't read pressure and IDs of any tire.
What's wrong? 
Is it my fault or is is it a (maybe known) issue with CanZE?

yoh-there, 2021-05-03:
We removed TMPS if the ZE50 is selected as we understand valve pairing etc can be done by the user now, so no CanZE needed anymore for that. So it's "by design".

Zoe-Fahrer, 2021-05-03:
Who did tell you, it can be done by the user? This would be nice, but it isn't.
It is still the same way, as it was on ZOE PI: you need the RENAULT garage or you have to use CanZE.
That's at least, I was told by RENAULT.
Can you advice me, how to do it without your great APP?

yoh-there, 2021-05-03:
I need to refer to the ZE50 drivers. Calling @Leobaillard @ik2wxx @FulminatingGhost

Leobaillard, 2021-05-03:
Hi!

This might help: https://www.speakev.com/threads/ze50-tpms-programming.158532/

The car can apparently learn about a second set of sensors. The system has to be reset by going to the TPMS screen on the dashboard and pressing "OK" until the values flash.

Zoe-Fahrer, 2021-05-03:
Thanks yoh-there adn Leobaillard!
I will let you know, if self learning of second sensor set works on my new ZE 50 / R135.

closed 2021-05-03 by Zoe-Fahrer