fesch/CanZE #698: Battery change, ph1 to ph1 battery

Erspange, 2021-05-04:
I have a Zoe with very poor battery health, and I have gotten hold of a newer battery. My current battery is 23KWh and has about 67% health, in the ZE app. When I change the battery, I need to set in the firmware the new capasity of the battery, is this possible in CANze? In case no, how do I go forward to change battery in the cars SW?

fesch, 2021-05-04:
The only thing that CanZE can modify / set in the car, are the TMPS sensors.
I don't think that it is possible to simply change the capacity of the battery somewhere. I rather think this is somewhere inside the BMS of the battery, so the new battery should already know about it.

Erspange, 2021-05-04:
Thanks for a quick answer!

This means I can just change battery and everything is ok...

fesch, 2021-05-04:
You mean "you by yourself" or "your dealer"?

Erspange, 2021-05-04:
I am a trained engineer, so more than capable of changing a battery ;)

fesch, 2021-05-04:
Ah, then I'm reassured :-)

closed 2021-05-04 by fesch