fesch/CanZE #699: Bug Canze avec Zoé Q210

StevenMeel, 2021-05-05:
Bonjour 
J'utilise depuis peu Canze
Je me connecte bien en wifi avec le konnwei
J'ai eu le soh
J'ai bien la température des modules 
Mais je n'arrive pas a avoir la tension des cellules, y a t il une astuce ?

[image]

[image]

ik2wxx, 2021-05-06:
Hello Steven.
Are you using the iOS version from https://testflight.apple.com/join/gypjWAtk ?
I guess you are using the iOS CanZE that's not mantained by the official CanZE Team (our screens are different).
I suggest you to try our app :-)

StevenMeel, 2021-05-06:
Oui, j'avais essayé celle qui était dans apple's app. 
Je vais essayer la vôtre au plus vite. 
merci

StevenMeel, 2021-05-06:
Effectivement avec votre application ça fonctionne. 
Merci

Pourquoi nous ne voyons que 96 cellules, nous en avons 192, non ?
[image]

ik2wxx, 2021-05-06:
Hello
Cells are coupled in pairs.
So there is no available tension control for every cell but only for coupled (1+1) cells.
Bye

closed 2021-05-06 by ik2wxx

StevenMeel, 2021-05-06:
Encore une question, j'accède au ECU LBC et pas au ECU LBC2. 
Est-ce normal ?
[image]
[image]