fesch/CanZE #701: ECU LCB2 qui s'affiche pad

StevenMeel, 2021-05-06:
Bonjour 
Juste une question, j'accède au ECU LBC et pas au ECU LBC2. 
Est-ce normal ?
[image]
[image]

StevenMeel, 2021-05-06:
Dommage car dans ces ECU LBC je vois la tension des 96 cellules , et j'aurais aimé avoir la tension des cellules des 96 autre dans LCB2.

yoh-there, 2021-05-06:
LBC2 is not "the other half". It is a backup computer system for the battery. LBC shows all voltages. The cells are paired, so a string of 96 pairs. Roberto already explained this in your issue #699. Closing.

closed 2021-05-06 by yoh-there