fesch/CanZE #702: Help for CANZE beginner

Mike2304, 2021-05-16:
I want to connect to my Zoe. I bought a Konnwei android dongle but it didn't work with my old Samsung tablet, so I bought a Konnwei Wifi dongle. The dongle seems to connect to my iphone, but I can't access the car. I get a message in French which says ELM Test "The wifi dongle was detected but it could not connect to it.
please disconnect the dongle, wait a few moments, reconnect it and try the operation again. make sure the vehicle is turned on".
I can't find any help on your website for new owners, so am requesting it here.
Regards Mike
2015 Zoe

yoh-there, 2021-05-16:
For Android, the Q&A https://canze.fisch.lu/qa/ might help, esp issue #9.

Calling @ik2wxx for iOS

ik2wxx, 2021-05-17:
Hello
Talking about iOS, that message is not from our official iOS app, maybe you're using another CanZE app that is not supported by us.
Please try the official iOS CanZE app currently available in Testflight at https://testflight.apple.com/join/gypjWAtk

Mike2304, 2021-05-17:
Hi,
Thanks, will do. Got app from Apple app store.
Mike

Mike2304, 2021-05-18:
Hello,
Thanks for reply,
I installed app to iphone from Test Flight. Successfully connected to
Konnwei and tested connection. Plug symbol shows on, but no data showing.
Car out of range of home wifi and mobile data off. Iphone shows wifi
connected to Konnwei.
Not sure if car is Q210, but is 2015 June. How can I tell? Feel I must be
doing something wrong.
Regards Mike

ik2wxx, 2021-05-19:
Hello Mike
In the iOS app, settings screen, please enable log and then, in the same screen, "test obd".
Wait for the test result, then open the File app, you will find the log of all communication between app and adapter.
You can post the log here, if the log content does not help you to understand the problem.
Roberto

Mike2304, 2021-05-24:
> Hello Mike
> Message related to "sd1" is coming from translations we are waiting to
> update (the app is in test).
> The File app is an Apple app available on every iOS device
> More info here https://support.apple.com/en-us/HT206481

Hi Robert,
I tried again without success. Nothing showing in "files".
To clarify:
Iphone is connecting to Konnwei and CanZe is showing connection to "Test
dongle" ok. All readings on CanZe are blank or zero, but it shows debugging
scrolling
CanZE settings are:
Device type ELM327
Connection _WiFi
Device address 192.168.0.10
device port 35000
Log to sdcard1 Yes
Write for emulator Yes
Regards Mike

ik2wxx, 2021-06-08:
Hello
Please try again with newer build 1.0 (6)
Thank you

Mike2304, 2021-06-08:
Hello,
Thanks for reply, we are now away from uk, so will try when we are back.
Regards Mike

Mike2304, 2021-08-08:
Hello,
Have tried this again today, still no results. I have 1(6)
[CanZE 1628436137.log](https://github.com/fesch/CanZE/files/6951123/CanZE.1628436137.log)
I am afraid I am completely unfamiliar with this and hope you can help. I just want to know my battery SOH because range is only 57 miles.
Regards Mike

ik2wxx, 2021-08-09:
> [CanZE 1628436137.log](https://github.com/fesch/CanZE/files/6951123/CanZE.1628436137.log)

Hi Mike
Reading from the log, I can see that connection between app and dongle is ok, connection between dongle and can bus is KO.
Most request have reply "CAN ERROR"
Also, request "atcra7ec" have reply "?", I suspect the dongle is not compatibile with Zoe.
Please try other obd dongle (or load a working one from a friend)
Supported obd dongles are listed here https://canze.fisch.lu/hardware/
BTW, I have a Konnwei 902 wifi (green one) and works great.

Mike2304, 2021-08-09:
Hi Roberto,
Thanks for your advice.
The first one I bought was Konnwei 902 BT (black), but it wouldnt work with
my old Samsung tablet and all my other devices are Apple.
So I then bought Konnwei 903 wifi (orange) which is this one for IOS.
I queried it with Konnwei and they sent the attached, which doesn't explain
what the difference is with the 902wifi.
Frustrating!
Mike

KarelSvo, 2021-08-15:
> Hi Roberto, Thanks for your advice. The first one I bought was Konnwei 902 BT (black), but it wouldnt work with my old Samsung tablet and all my other devices are Apple. So I then bought Konnwei 903 [...]

to understand how counterfeiting is done   https://cvtz50.info/en/elm327/
an example for 4mhz elm 
aliexpress.com/item/4000159235000.html
ebay.com/itm/123997779408
kingbolentool.com/collections/basic-obd2-code-reader/products/kingbolen-obd2-scanner-bluetooth-check-engine-light?variant=31540043481145

ik2wxx, 2021-09-06:
> I queried it with Konnwei and they sent the attached, which doesn't explain what the difference is with the 902wifi. 

Maybe you forgot to attach a file or an image ?

mkozachok, 2022-07-23:
I have the same issue. Is it competitive with Konnwei 901 WIFI?

closed 2023-10-02 by yoh-there