fesch/CanZE #703: canZE on Twingo ZE

filippoagresti, 2021-05-21:
Hi!
Thank you for developing this nice app. I own a Twingo ZE and I can partially use the app setting as ZOE R90/110 but not as Twingo 3 Ph2. Anyway, I can get many useful parameters for example during charging. I could help to test new versions of the app on my Twingo and provide information!

yoh-there, 2021-06-05:
Thank you for your offer for testing. None of the current developers does.  Experience from the past dictates trying to develop "in the blind" is only leading to tears.

However, if anyone owning one, has an interest in car networking **and** is happy to do some Android Studio coding **and** some long cold days with the vehicle, it could always happen. The code is made in a way it could fairly easily fit in.

Sorry for not having a sunnier answer.

closed 2021-06-15 by yoh-there

asez73, 2022-01-26:
Hi, I use, on my Twingo ZE, this very same dongle and CanZE with the Zoe R90/110 and got really good results with almost everything: charging curves, power delivery from charging station and battery charging power included.
Except for certains specific screens which I get from the Zoe RZE40/ZE50: 
-battery modules voltages and temperatures from their respective detailled  screens,
-from the screen Historiques des charges i get the 4 lines below charging history, the historique is really obtained from Zoe R90/110 setup, not the last 4 lines...

So I was wondering, as I failed to install Android Studio which doesn't load on windows 11 (that's the emulator apparently), I you could create a specific profile for the Twingo ZE based on those informations? Screenshots and logs might help for this too.