fesch/CanZE #704: Problème de charge/compréhension application CanZE

Keulkeul10, 2021-05-31:
Bonjour, 
Je me présente, Michaël .
Je suis propriétaire d'une Renault ZOE et j'utilise l'application CanZE que je trouve vraiment complète.

Cependant j'ai quelques questions concernant son utilisation. 
Je vais essayer d'être le plus clair possible.

J'ai des problèmes de charge, et je souhaiterais avoir des éclaircissements sur certains points. 

Pendant la charge, sur l'application, lorsque je lance la page TECHNIQUE puis CHARGE je retrouve ma valeur de terre mesurée par le véhicule qui est plutôt bonne et 2 lignes en dessous il y a ÉTAT D'AVANCEMENT avec le message FUITE RES. HF.
À quoi correspond ce message ? Quel est sa signification ? 

Ma seconde question concerne les défauts dans LECTURE DTC. 
Pour trouver mon code défaut en lien avec ma charge je vais sur la page BCB-OBC (j'ai trouvé cela sur des forums pour la page à consulter). 
Mon souci c'est que j'ai l'impression que l'interrogation se fait mais que j'ai toujours le même ordre de message d'écrit. 
Comment savoir quel est le dernier code apparu ? 
S'agit il du premier de la liste ? 

Si vous prenez le temps de me lire et de me répondre je pense que j'en saurais déjà plus sur mon problème. 
Par avance merci à vous. 

Cordialement.

Keulkeul10, 2021-05-31:
Merci Google traduction

Hello,
Let me introduce myself, Michaël.
I own a Renault ZOE and I use the CanZE app which I find really complete.

However I have a few questions regarding its use.
I will try to be as clear as possible.

I have charging problems, and I would like some clarification on certain points.

While charging, on the application, when I launch the TECHNICAL then CHARGE page, I find my earth value measured by the vehicle which is rather good and 2 lines below there is PROGRESS STATE with the message RES LEAK. HF.
What is this message for? What is its significance?

My second question concerns the flaws in READ DTC.
To find my fault code related to my load, I go to the BCB-OBC page (I found this on forums for the page to consult).
My concern is that I have the impression that the interrogation is done but that I always have the same order of the written message.
How do I know what the last code appeared?
Is this the first on the list?

If you take the time to read me and answer me I think I would already know more about my problem.
Thank you in advance.

Cordially.

Shrolk, 2021-06-02:
Salut Michaël,

Concernant ton problème de charge tu es sur quelle génération de Zoé et quel moteur ?
Tu recharges sur une prise domestique classique ?

Au plaisir

Keulkeul10, 2021-06-02:
Bonjour,
tout d'abord merci d'avoir pris le temps de lire mon message.
Il s'agit d'une ZOE LIFE de 2013 avec un moteur Q210.
Mes problèmes arrivent avec le chargeur domestique sur prise classique.
L'essai d'inversion phase/neutre a été fait, la prise en question est sur disjoncteur 20A et un différentiel rien qu'à lui seul.
J'ai pu recharger chez l'ancien propriétaire qui est proche de chez moi, il n'y a aucun problème.
La terre était vue à 13 ohm lors du diagnostic électrique de la maison en juin 2020 et lorsque la charge se lance, la mesure de terre sur l'appli CANZe varie entre 7 et 40 ohm.
La charge se lance très rapidement lors de la vérification. Mais parfois au bout d'une heure elle va stopper, ou alors au bout de 10 minutes. C'est vraiment aléatoire.
Je ne sais pas si je suis allé trop loin dans mes explications mais j'essaie d'être le plus précis possible.

Dans l'attente de vous lire.

yoh-there, 2021-06-02:
40 ohms ground resistance is not a problem. The old Q210 (which I own too) shut off at 100 ohm with the old firmware, 150 with the new firmware. If you want to know which firmware you have, go to the firmware screen and tap LBC. The new one is 0854 in the Soft line or higher. So I doubt this is an earth resistance problem. However, if yours is on lower firmware, I would advise you to ask the dealer for firmware updates for the LBC and BCB.

HF leak is the leak current to earth at roughly 10 kHz (the chargers frequency). Check the LEAK CURRENTS screen while you charge. Mine is around 40-45 mA in the HF graph. If (if!!) this is the problem, it could be a bad insulation or sensor problem in your car and I think the only solution is to visit the dealer.

Disclaimer: I am not a Renault mechanic, just have in interest in my car

Shrolk, 2021-06-02:
Alors le mieux sera que tu installes une prise type GreenUP. Renault ne te fera pas d'intervention sans cette prise

Keulkeul10, 2021-06-02:
yoh-there, thank you for your answer.
The HF leakage current is generally around 200 mA.
As an example the attached captures.
For firmware i can't answer you tonight i'm on the go until next week.
Thanks
[image]
[image]

Keulkeul10, 2021-06-02:
Shrolk, merci de ta réponse. Franchement si je pouvais me passer de ce type de prise ça m'arrangerait.
Quand je recharge, je ne suis pas pressé et j'ai bien peur que mon problème soit répercuté même avec une GreenUp.
Je pense ne pas être le seul à recharger sur prise classique.
Merci de ta réponse en tout cas.

Shrolk, 2021-06-02:
> Shrolk, merci de ta réponse. Franchement si je pouvais me passer de ce type de prise ça m'arrangerait.
> Quand je recharge, je ne suis pas pressé et j'ai bien peur que mon problème soit répercuté même [...]

J'ai eu le même problème que toi sur ma Zoé Q210 de 2014. Impossible de charger sur prise classique (deux changements du circuit de charge de la Zoé aux frais du garage) mais pas de soucis quand je suis sur GreenUp

Keulkeul10, 2021-06-02:
> J'ai eu le même problème que toi sur ma Zoé Q210 de 2014. Impossible de charger sur prise classique (deux changements du circuit de charge de la Zoé aux frais du garage) mais pas de soucis quand je [...]

J'ai du mal à comprendre comment la Green Up peut corriger un problème d'harmoniques si c'est mon cas.
Je sais qu'elle autorise un "tirage" plus élevé et est renforcée pour le courant qui est plus élevé passant par ses bornes.
Au pire si je n'ai toujours pas su régler mon problème, je peux récupérer cet été une prise d'un ancien propriétaire qui l'a toujours d'installée chez lui.

Shrolk, 2021-06-07:
> J'ai du mal à comprendre comment la Green Up peut corriger un problème d'harmoniques si c'est mon cas.
> Je sais qu'elle autorise un "tirage" plus élevé et est renforcée pour le courant qui est plus [...]

Le fait de tirer plus fort augmente le rendement du chargeur et donc réduit sa génération d'harmoniques

Keulkeul10, 2021-06-07:
> Le fait de tirer plus fort augmente le rendement du chargeur et donc réduit sa génération d'harmoniques

Ah d'accord, je ne savais pas, merci pour l'info.
J'ai acheté un filtre anti-harmonique, je vais voir pour l'installer dans la semaine sur le ligne de ma prise, à voir si ça corrige le problème.
Sinon, bah greenup.... Merci pour toutes ces infos.

yoh-there, 2021-08-29:
Closing as this issue is stale and not related to CanZE anyway. Thanks for the contributions.

closed 2021-08-29 by yoh-there