fesch/CanZE #706: Latest version misreading values

sirdir1972, 2021-06-05:
Hi

The latest beta seems to read values more quickly from the ODB adapter. Problem is it misreads most values.

yoh-there, 2021-06-05:
Thanks for reporting. Version 1.60, the latest,  was released on februari 4th. Are you sure there is this causality? If so, can you please provide us with as much information as possible?
- car type (esp if it's a Ph1 or Ph2 car; if it's a Ph1, please try the "ISOTP" setting)
- dongle used
- settings

sirdir1972, 2021-06-05:
Yes, I'm talking about the iOS beta 1.0 (6)

The Dongle is a LELink BLE btw, the car a Zoe Q210

yoh-there, 2021-06-05:
Got it, re-assigned. BTW, I am not sure if the ISOTP setting is in the iOS version, trying that might help.

sirdir1972, 2021-06-05:
ISOTP is active

sirdir1972, 2021-06-08:
Sure, here are some screenshots. It's coincidence what data is correct and what not and it may change with every 'loop' of querying the ODB, I'd say:
https://www.icloud.com/sharedalbum/#B0v5M7GFPxybc1

ik2wxx, 2021-06-10:
Thank you.
As I'm reading from screenshots, data from car is not recognized.
Data is NAN (not a number) or EMPTY when no reply received.
Is the car model correct ?
Most Zoe models need specific command to read data and settings wrong car model will prevent you to receive correct data.
I can't say this is the problem solution, I'm trying to help.
Also, can you please enable logs and retry to fetch some data, than share logs with us ?
Logs can be enabled in settings screen, and are written on your iphone, you can find them with Apple File app under "CanZE" folder.
Apple File app can be downloaded from Apple App Store if you don't have it already
https://apps.apple.com/app/file/id1232058109

sirdir1972, 2021-06-10:
Hi!
Car Model is correct (Q210) and it has all worked until I updated to the latest beta. It seems to read the data quicker than before but not always correctly.

I’ll attach the logs

Regards
Patrick

ik2wxx, 2021-06-11:
Thank you for reporting, we will investigate.
Build 5 is expired so you can't install it anymore, sorry.

Any other Q210 or Lelink user, please tell us if you have problems.

wolfijenne, 2021-06-24:
Just tried it out (same configuration with R90). I made a video of it, maybe it helps.

https://www.spider.li/RPReplay_Final1624102662.MP4

ik2wxx, 2021-06-28:
Thank you, can you please also post Settings screen ?

> Just tried it out (same configuration with R90). I made a video of it, maybe it helps.
> 
> https://www.spider.li/RPReplay_Final1624102662.MP4

Giankymis, 2021-08-20:
Hello, i have a wifi dongle. Is there anyone that have configured It over http? I have try to add dongle ip address but not works. Thanks

wolfijenne, 2021-08-20:
sorry, did not see this earlier. 
[image]

sirdir1972, 2021-09-05:
any news on this? Still can't use CanZe…

ik2wxx, 2021-09-06:
> any news on this? Still can't use CanZe…

We're waiting for report of other Q210 users...but looks like other Q210 have no issues.
Did you tried changing OBD adapter with any of the supported model ?
BTW, I did not receive your logs.

wolfijenne, 2021-09-06:
I also have this problem. I would love to send you logs, however, I have no idea where to find them on my iphone as the iphone.

thx, Wolfgang

ik2wxx, 2021-09-06:
> I also have this problem. I would love to send you logs, however, I have no idea where to find them on my iphone as the iphone. thx, Wolfgang

Your car model is R90, not Q210, let's separate issues.

Logs can be enabled in settings screen, and are written on your iphone, you can find them with Apple File app under "CanZE" folder.
Apple File app can be downloaded from Apple App Store if you don't have it already
https://apps.apple.com/app/file/id1232058109

wolfijenne, 2021-09-06:
Thx... I finally found them. As I tried it just recently, here is a log. The log should consist of the charging screen and the cell Voltages screen. If you want me to try something specifically, pls let me know.

[Emulator 1630791445.log](https://github.com/fesch/CanZE/files/7116475/Emulator.1630791445.log)
[CanZE 1630791439.log](https://github.com/fesch/CanZE/files/7116477/CanZE.1630791439.log)

ik2wxx, 2021-09-06:
> Thx... I finally found them. As I tried it just recently, here is a log. The log should consist of the charging screen and the cell Voltages screen. If you want me to try something specifically, pls [...]

Thank you for reporting log files, let me check.

sirdir1972, 2021-09-06:
> > any news on this? Still can't use CanZe…
> 
> We're waiting for report of other Q210 users...but looks like other Q210 have no issues.
> BTW, I did not receive your logs, can you please resend them ?

Sure, I'll send you the logs I currently have. 
I'm not sure he Q90 and Q210 issues are different, they look very similar. I don't have access to another compatible blutooth dongle, but this one worked fine before the update. 
[CanZE 1630934434.log](https://github.com/fesch/CanZE/files/7117549/CanZE.1630934434.log)

sirdir1972, 2022-01-04:
Bought another BLE dongle. Works fine with ABRP but same problem with CanZE. I have the impression CanZE communicates faster with the car/dongle than it did before. Maybe it would help to add some delay or a lower baudrate or something?

jakub-a, 2022-03-24:
Here is Q90 ZE40 (2019). Using the Vgate iCar Pro BT4.0. Values are not, or not read correctly:
[image]

[image]

https://user-images.githubusercontent.com/67999691/159875019-b15be840-45e0-4348-8449-cc2b598775ec.MP4

ik2wxx, 2022-03-24:
> Here is Q90 ZE40 (2019). Using the Vgate iCar Pro BT4.0 and CanZE 1.0 (9). Values are not available or not read correctly: 

Hi, did you try other car models in setting screen?
Try both "ZOE ZE40/ZE50" and "ZOE R90/110" and "ZOE Q90" to find your correct setting.

jakub-a, 2022-03-26:
I tried all models but I am not getting the correct readings or no readings at all.

ik2wxx, 2022-03-28:
> I tried all models but I am not getting the correct readings or no readings at all.

Then the problem seems to be the dongle...can you try a different one?

jakub-a, 2022-03-28:
Good to know. Thank you - I will return it then. Unfortunately I don't have a second iCar Pro. I do have a BLE dongle that works on my old android device but the iOS App only allows 2 devices: Vgate iCar Pro and LELink.
Is there a way to pair with a different device?

Here is the iCar Pro that doesn't seem to work.
[image]
.

ik2wxx, 2022-03-28:
> Good to know. Thank you - I will return it then. Unfortunately I don't have a second iCar Pro. I do have a BLE dongle that works on my old android device but the iOS App only allows 2 devices: Vgate iCar Pro and LELink. Is there a way to pair with a different device?

Go to the last page scrolling the home view to the left, you will see a "TEST OBD" button.
Click "connect" and select the OBD BLE name, then try with "Auto Init" and "TEST" button to see if it works.
If it works, send me then OBD BLE name and I will add it as compatible.

jakub-a, 2022-03-28:
I connected to the dongle both the iCar Pro as well as the other BLE Dongle but the App Crashes every time i try to run a rest. 

Happy to help by sending logs and whatnot. 

https://user-images.githubusercontent.com/67999691/160434486-b54dec87-ca91-4852-a87f-df22ba9eede2.MOV

jakub-a, 2022-03-28:
I did it as you instructed. Still the test button crashes the app. 
https://user-images.githubusercontent.com/67999691/160490222-73021d79-956e-4a57-8628-729041c04ec0.MOV

jakub-a, 2022-03-28:
https://user-images.githubusercontent.com/67999691/160490650-455974b4-3b40-4709-8635-7fc8bcfe9359.MOV

ik2wxx, 2022-03-29:
> I did it as you instructed. Still the test button crashes the app.

I will add the OBDII ble device in the settings screen so you can test it with the full app functions.
Wait for Testflight alert to update the app.
Let us know!

jakub-a, 2022-03-29:
Tested both dongles: the iCar Pro and the unknown branże OBDII BLE. Both connect but both read give wrong readings for some parameters. But compared to the build 9 more have correct values. In this and previous versions they are slow to update values. Voltage heat map crashed the app. The build 11 also caused the dashboard to turn on and a message about automatic lock was shown briefly. Should I be logging these sessions for you?
[image]
[image]

ik2wxx, 2022-03-30:
Thank you for reporting, we always have issues with ELM327 dongles and I don't have any experience with cars that are not model ZE50.
I need to check the app.
Will come back as I have any news.

jakub-a, 2022-03-30:
This ELM327 dongle has been tested on Android with CanZE and is working fine. It looks like it's dongle independed because it shows similar errors for both the no-name ELM327 as well as iCar Pro BLE.
If it was working before (as other users report) on the Q model, then something must have changed how the app communicates to cause this regression.

jakub-a, 2022-03-31:
> Thank you for reporting, we always have issues with ELM327 dongles and I don't have any experience with cars that are not model ZE50.
> I need to check the app.
> Will come back as I have any news.

@ik2wxx  - in one of the previous posts you suggested the iCar Pro dongle is defective. Do you still maintain that it's broken given more evidence?

ik2wxx, 2022-03-31:
No, I've tried by myself and I think the problem can be solved by the app.
Communication between app and dongle is a real pain (mainly for data receiving).
I've tried slowing the dongle command queue and now it's better, but not a real problem solution.
BTW I will publish a version with adjustable dongle command queue delay so you can experiment on your car.
I'm using a value of 0.20 in my ZE50 and works ok.
Please note that a slower delay will make the data receiving faster, but you can receive wrong data.
Also I noticed that, after a wrong data is received, all other data are not fully reliable!!! Using a slower delay will not solve the problem, you will have to disconnect the dongle from the car, reconnect to the car, reconnect to the app :-(
I hope to release the new version today.

sirdir1972, 2022-03-31:
Doesn't work, even on 0.50. the temperatures in the advanced charging screen get messed up every time. 
It still works with build 5. Build 6 and onwards breaks it. 
BTW what's about the 'energy to full' value, that's always messed up. Does work on the Android version.

jakub-a, 2022-04-04:
Is there any way I can use the **build 5** @sirdir1972 @ik2wxx ?

No delay setting works properly. It is starting to receive meaningful data at a delay=0.03s but increasing it doesn't really improve things much. I have it at 0.05 right now. There are some data being received properly such as `pedal position` or `wheel torque` (with occasional hiccups).

Thanks for trying to fix it!

ik2wxx, 2022-04-04:
Rolling back to older releases will delete latest updates (translations, fixes, features).
Please wait until I can add those news into the older release, without touching connection with ELM.

derykl, 2023-03-02:
Seeing similar here with vgate icar pro ble 4.0.  on the default delay of 0.2 it "kind of works on some screens" - refuses to report max charge/Regen, real state of charge, only one cell is read for temperature (and is wrong) etc.   Using 1.0.1 from TestFlight. 

If I connect the same dongle to my Android phone which seems to use standard Bluetooth and need pairing, everything works quickly and accurately.  Problem is specific to ios / BLE

sirdir1972, 2023-03-02:
>  Problem is specific to ios / BLE

Yep, but still works in an older version. Unfortunately I'm having problems getting it compiled (managed it a while ago, but deleted my phone and would have to reinstall the older version again)

ik2wxx, 2023-03-03:
> Seeing similar here with vgate icar pro ble 4.0. on the default delay of 0.2 it "kind of works on some screens" - refuses to report max charge/Regen, real state of charge, only one cell is read for [...]

iOS version need some fixes for BLE adapters, I'm sorry I didn't find any time to work on it. That's why the app is still in development.

ik2wxx, 2023-03-03:
> > Problem is specific to ios / BLE
> 
> Yep, but still works in an older version. Unfortunately I'm having problems getting it compiled (managed it a while ago, but deleted my phone and would have to reinstall the older version again)

What's the version that works better for you ?

sirdir1972, 2023-03-03:
> What's the version that works better for you ?

Huh, it's so long now.. The version that was out on June 5th 2021 already had the problem. The version before was fine. 1.0(5) was still OK? Would that match the releases? Reread the thread and yes, version 5 was still OK.

derykl, 2023-03-03:
> > Seeing similar here with vgate icar pro ble 4.0. on the default delay of 0.2 it "kind of works on some screens" - refuses to report max charge/Regen, real state of charge, only one cell is read for temperature (and is wrong) etc. Using 1.0.1 from TestFlight.
> > If I connect the same dongle to my Android phone which seems to use standard Bluetooth and need pairing, everything works quickly and accurately. Problem is specific to ios / BLE
> 
> iOS version need some fixes for BLE adapters, I'm sorry I didn't find any time to work on it. That's why the app is still in development. Also I don't have the car always available for tests :-(

Thanks in any case for your work on it.  It does the minimum I need it to do which is show charging rate

If I can help in some way let me know

ik2wxx, 2023-10-10:
This should have been solved with last version

sirdir1972, 2023-10-16:
Hi!

It’s not, i.E. battery temperatures are still mostly read as NaN

ik2wxx, 2023-10-19:
> Hi! It’s not, i.E. battery temperatures are still mostly read as NaN

Hello
All battery values, cell temperature and cell voltage included, are correctly read with the app.
I used a VGate OBD dongle.

closed 2023-10-30 by ik2wxx