fesch/CanZE #709: ELM version 2.x

ZMan88, 2021-06-24:
It works :)
[image]
And the dongle is:
[image]
Car: Zoe ph2 R135
Hope this helps.

yoh-there, 2021-06-24:
Thanks for letting us know! I'll need to think on what to do. The 2.0 version as reported by the dongles is almost always "made up versioning" by non working clones. Seems iCarPro either has a really good clone of the true ELM or HAS a true ELM. Maybe just change the wording of the message. Thanks again. I'll leave the issue open until code has changed (in whatever way ;-) )

ZMan88, 2021-06-24:
Running ELM Identifier app I have the following result:
Scan date=2021-06-24 19:33:09
Device name=Android-Vlink
Device mac=...
Device Version (Declared)=ELM327 v2.2

Command	Version	Result	RAW
AT@1	1.0	OK	AT@1\rOBDII to RS232 Interpreter\r\r>	
ATSPA6	1.0	OK	ATSPA6\rOK\r\r>	
ATTPA6	1.0	OK	ATTPA6\rOK\r\r>	
ATTP6	1.0	OK	ATTP6\rOK\r\r>	
ATSP00	1.3	OK	ATSP00\rOK\r\r>	
ATSP6	1.0	OK	ATSP6\rOK\r\r>	
ATAL	1.0	OK	ATAL\rOK\r\r>	
ATNL	1.0	OK	ATNL\rOK\r\r>	
ATAMC	2.0	OK	ATAMC\r02\r\r>	
ATAMT20	2.0	OK	ATAMT20\rOK\r\r>	
ATRA70	1.3	OK	ATRA70\rOK\r\r>	
ATSR70	1.2	OK	ATSR70\rOK\r\r>	
ATAR	1.2	OK	ATAR\rOK\r\r>	
ATAT0	1.2	OK	ATAT0\rOK\r\r>	
ATAT2	1.2	OK	ATAT2\rOK\r\r>	
ATAT1	1.2	OK	ATAT1\rOK\r\r>	
ATBD	1.0	OK	ATBD\r00 18 DA F1 DA 06 50 03 00 32 01 F4 AA \r\r>	
ATBRT0F	1.2	OK	ATBRT0F\rOK\r\r>	
ATCAF0	1.0	OK	ATCAF0\rOK\r\r>	
ATCAF1	1.0	OK	ATCAF1\rOK\r\r>	
ATCEA04	1.4	OK	ATCEA04\rOK\r\r>	
ATCERF1	2.2	OK	ATCERF1\rOK\r\r>	
ATCEA	1.4	OK	ATCEA\rOK\r\r>	
ATCF00000111	1.0	OK	ATCF00000111\rOK\r\r>	
ATCF111	1.0	OK	ATCF111\rOK\r\r>	
ATCFC0	1.0	OK	ATCFC0\rOK\r\r>	
ATCFC1	1.0	OK	ATCFC1\rOK\r\r>	
ATCM00000111	1.0	OK	ATCM00000111\rOK\r\r>	
ATCM111	1.0	OK	ATCM111\rOK\r\r>	
ATCP18	1.0	OK	ATCP18\rOK\r\r>	
ATCRA7C0	1.3	OK	ATCRA7C0\rOK\r\r>	
ATCRA000007C0	1.3	OK	ATCRA000007C0\rOK\r\r>	
ATCRA	1.4b	OK	ATCRA\rOK\r\r>	
ATCS	1.0	OK	ATCS\rT:00 R:00 F:0\r\r>	
ATCSM0	1.4b	OK	ATCSM0\rOK\r\r>	
ATCSM1	1.4b	OK	ATCSM1\rOK\r\r>	
ATCTM5	2.1	OK	ATCTM5\rOK\r\r>	
ATCTM1	2.1	OK	ATCTM1\rOK\r\r>	
ATCV0000	1.4	OK	ATCV0000\rOK\r\r>	
ATD1	1.3	OK	ATD1\rOK\r\r>	
ATD0	1.3	OK	ATD0\rOK\r\r>	
ATDP	1.0	OK	ATDP\rISO 15765-4 (CAN 11/500)\r\r>	
ATDPN	1.0	OK	ATDPN\r6\r\r>	
ATE0	1.0	OK	ATE0\rOK\r\r>	
ATE1	1.0	OK	OK\r\r>	
ATFCSD0430FF00	1.1	OK	ATFCSD0430FF00\rOK\r\r>	
ATFCSH000007B0	1.1	OK	ATFCSH000007B0\rOK\r\r>	
ATFCSH7B0	1.1	OK	ATFCSH7B0\rOK\r\r>	
ATFCSM0	1.1	OK	ATFCSM0\rOK\r\r>	
ATFE	1.3a	OK	ATFE\rOK\r\r>	
ATH0	1.0	OK	ATH0\rOK\r\r>	
ATH1	1.0	OK	ATH1\rOK\r\r>	
ATI	1.0	OK	ATI\rELM327 v2.2\r\r>	
ATIB10	1.0	OK	ATIB10\rOK\r\r>	
ATIB12	2.2	OK	ATIB12\rOK\r\r>	
ATIB15	2.2	OK	ATIB15\rOK\r\r>	
ATIB48	1.4	OK	ATIB48\rOK\r\r>	
ATIB96	1.0	OK	ATIB96\rOK\r\r>	
ATIFRS	1.2	OK	ATIFRS\rOK\r\r>	
ATIFRH	1.2	OK	ATIFRH\rOK\r\r>	
ATIFR0	1.2	OK	ATIFR0\rOK\r\r>	
ATIFR2	1.2	OK	ATIFR2\rOK\r\r>	
ATIFR1	1.2	OK	ATIFR1\rOK\r\r>	
ATIFR6	2.2	OK	ATIFR6\rOK\r\r>	
ATIFR5	2.2	OK	ATIFR5\rOK\r\r>	
ATIFR4	2.2	OK	ATIFR4\rOK\r\r>	
ATIGN	1.4	OK	ATIGN\rON\r\r>	
ATIIA7A	1.2	OK	ATIIA7A\rOK\r\r>	
ATJS	1.3	OK	ATJS\rOK\r\r>	
ATJE	1.3	OK	ATJE\rOK\r\r>	
ATJHF0	1.4b	OK	ATJHF0\rOK\r\r>	
ATJHF1	1.4b	OK	ATJHF1\rOK\r\r>	
ATJTM5	1.4b	OK	ATJTM5\rOK\r\r>	
ATJTM1	1.4b	OK	ATJTM1\rOK\r\r>	
ATKW	1.3	OK	ATKW\r1:-- 2:--\r\r>	
ATKW0	1.2	OK	ATKW0\rOK\r\r>	
ATKW1	1.2	OK	ATKW1\rOK\r\r>	
ATL0	1.0	OK	ATL0\rOK\r\r>	
ATL1	1.0	OK	ATL1\rOK\r\n\r\n>	
ATM1	1.0	OK	ATM1\r\nOK\r\n\r\n>	
ATM0	1.0	OK	ATM0\r\nOK\r\n\r\n>	
ATPPFFON	1.1	OK	ATPPFFON\r\nOK\r\n\r\n>	
ATPPFFOFF	1.1	OK	ATPPFFOFF\r\nOK\r\n\r\n>	
ATPPS	1.1	OK	ATPPS\r\n00:FF F  01:FF F  02:FF F  03:32 F\r\n04:01 F  05:FF F  06:F1 F  07:09 F\r\n08:FF F  09:00 F  0A:0A F  0B:FF F\r\n0C:68 F  0D:0D F  0E:FA F  0F:D5 F\r\n10:0D F  11:00 F  12:FF F  13:55 F\r\n14:50 F  15:0A F  16:FF F  17:92 F\r\n18:31 F  19:4F F  1A:0A F  1B:0A F\r\n1C:03 F  1D:0F F  1E:4A F  1F:FF F\r\n20:FF F  21:FF F  22:FF F  23:FF F\r\n24:00 F  25:00 F  26:00 F  27:FF F\r\n28:FF F  29:FF F  2A:3C F  2B:02 F\r\n2C:E0 F  2D:04 F  2E:80 F  2F:0A F\r\n\r\n>	
ATR0	1.0	OK	ATR0\r\nOK\r\n\r\n>	
ATR1	1.0	OK	ATR1\r\nOK\r\n\r\n>	
ATRV	1.0	OK	ATRV\r\n12.3V\r\n\r\n>	
ATS0	1.3	OK	ATS0\r\nOK\r\n\r\n>	
ATS1	1.3	OK	ATS1\r\nOK\r\n\r\n>	
ATSDFF	1.4	OK	ATSDFF\r\nOK\r\n\r\n>	
ATRD	1.4	OK	ATRD\r\nFF \r\n\r\n>	
ATSH000007E0	2.0	OK	ATSH000007E0\r\nOK\r\n\r\n>	
ATSHA848F1	1.0	OK	ATSHA848F1\r\nOK\r\n\r\n>	
ATSH7B0	1.0	OK	ATSH7B0\r\nOK\r\n\r\n>	
ATST32	1.0	OK	ATST32\r\nOK\r\n\r\n>	
ATSWFF	1.0	OK	ATSWFF\r\nOK\r\n\r\n>	
ATTAF9	1.4	OK	ATTAF9\r\nOK\r\n\r\n>	
ATV1	1.3	OK	ATV1\r\nOK\r\n\r\n>	
ATV0	1.3	OK	ATV0\r\nOK\r\n\r\n>	
ATWMC133F13E	1.0	OK	ATWMC133F13E\r\nOK\r\n\r\n>	
ATPC	1.0	OK	ATPC\r\nOK\r\n\r\n>	
ATWS	1.0	OK	ATWS\r\n\r\rELM327 v2.2\r\r>	
ATD	1.0	OK	ATD\r\rOK\r\r>	
ATZ	1.0	OK	ATZ\r\r\rELM327 v2.2\r\r>

closed 2023-10-02 by yoh-there