fesch/CanZE #711: Zoe Q210 (2015) charging power oscillations at 43 kW AC

madamic, 2021-08-14:
Hello there!

My Zoe Q210 (2015) has been acting strange on 43 kW fast chargers in the past month. I have "available charging power" oscillations from 20 kW to 42 kW, with a period of cca. 1 minute. The car charges with 20 kW, and then slowly and linearly ramps the power up until it reaches 42 kW. It holds maximum power for around 10 seconds, and then drops immediately back down to 20 kW. Then, the cycle repeats. So if you imagine it, my charging power curve looks like a sawtooth wave signal between 20 and 42 kW. When the battery is full enough so that the "max battery charge" value drops down to 30 kW, the charger can hold 30 kW continuously without any oscillations.

The "Charge pilot Amp" value is always at 64 A, and the same issue appears at different charge points, so the problem is definitely in the car. The cooling systems are all running as they should. The coolant level is OK (at Max) and it's circulating, and the fan is spinning at 90%. The AC compressor runs smoothly and battery cooling works as it should, so the cell modules never exceed 34 °C. There is enough gas and the pipes are cold. The outside temperature of the test was 25°C, rising towards 30°C.

I'm suspecting the power electronics in the BCB get too hot at 42 kW, so the power is reduced, the temperature drops, and the cycle repeats. But it's never done this before, this is the first time (I think, but not 100% sure). 2 months ago, I have had firmware updates of the EVC, BCB, BMS & BMS-Safety computers to the latest versions. Is it possible that the new firmware for BCB has a lower Max Temp threshold for the IGBTs and therefore does not allow full continuous 43 kW in summer temps? I will have to repeat the test in colder weather, but I don't have this option now. :D

If I read DTCs from the "BCB-OBC" while fast charging, the error which immediately catches my attention is:
DTC0631fb (P031fb) - "Electronic power driver temperature" with flags "tstFailThisOp, confirmedDtc, failedSinceClear".

Has anyone seen this behavior before or has any idea what could be going on? Otherwise, the car works great, 22 kW charging works normally, aircon works very well, there are no warning lights on the dash etc.

Thank you very much for your help!

fesch, 2021-08-17:
If you want, you may try to set the car in ECO-Mode before charging it.
It has been demonstrated that having the car in ECO-Mode does affect the charging behavior ... at least if it is warm outside ;)

madamic, 2021-08-17:
Good news! I tried again today with lower outside temperatures (15°C), so more than 10 degrees less than last time. Everything worked as it should, no oscillations, constant charging power at 40 kW AC. Also, the "Electronic power driver temperature" DTC had only flags "confirmedDtc, failedSinceClear" active, so without "tstFailThisOp".

So it seems the problem last time was indeed overheating! The question remains, why did it overheat at all at "pleasant" 25 °C, which is not scorching hot.

yoh-there, 2021-08-29:
Happy it seems better. A bit strange as the Q210 charges 40 kW while the battery is >35C. I suggest a (dealer) check of the cooling circuit.

Closing as it has nothing to do with CanZE.

closed 2021-08-29 by yoh-there