fesch/CanZE #713: RFE: in cells voltage display some statistics like minimum and maximum

muayyad-alsadi, 2021-08-22:
Hello,

I would like to thank you for this useful software

it would be nice to report minimum and maximum cells voltage (of the high voltage battery) in the cells voltage graph
because if those two reading are close to gather it's a good indicator, if there is a wide gap between the two it's a bad indicator

muayyad-alsadi, 2021-08-22:
[image]
[image]

muayyad-alsadi, 2021-08-22:
while looking at the code, it seems already report min/max

https://github.com/fesch/CanZE/commit/b4024b5ede57ab4d7ad236132dd81597911fe310

was this commit released google play?

FulminatingGhost, 2021-08-22:
Hi,
You can see minimum and maximum voltage in the cell voltage heatmap view.

muayyad-alsadi, 2021-08-22:
Can we add it to the graph?

fesch, 2021-08-22:
I have updated the plotter widget accordingly to this. The next release will display min / max.

closed 2021-08-22 by fesch

muayyad-alsadi, 2021-08-22:
Thank you very much