fesch/CanZE #715: SOH (Konnwei working)

Mike2304, 2021-08-29:
Hi,
Have got CanZE working, and SOH reads 71%. Before I go and pay for an official check - is this likely to be accurate? This is reading from the 'charging' screen on CanZE, the 'battery' screen doesn't show a number.
Regards Mike

yoh-there, 2021-08-29:
71% is too low. My 2013 Q210 is on 83%.

What to do depends on what car we are talking about. Assuming a somewhat older model given the SOH, you need to check the firmware of the LBC. Field "Soft" should be on 0854 or higher. I bet it's not. If I am correct, visit the dealer and say: "the range is really not up to snuff with what other driver of that vintage report. And when I am charging it's hanging for hours on 99%". That might get you the firmware update for free as it's a know issue.

Assuming again they do the firmware update for you (paid for or not), you will find the SOH at 98%. It will take months for it to trickle down to the "real" level.

Closing as it has nothing to do with CanZE ;-)

closed 2021-08-29 by yoh-there

ik2wxx, 2021-08-30:
Hi Mike.
I asked to Renault local workshop to check SOH when my ZE50 was one year old.
CanZE reading was 81% and Renault reading was 81%.
So, apart from the erroneous reading due to a LBC old firmware, the reading of CanZE was identical to the Renault workshop.

Mike2304, 2021-08-30:
Thanks Roberto

Mike2304, 2021-08-30:
Thanks for reply. Checked LBC Field soft is 751.
Mike