fesch/CanZE #716: Some connection issues with ELM327

Flodesirat, 2021-09-01:
Hello,

I bought an ELM327 Bluetooth on amazon: https://www.amazon.fr/gp/product/B093T5V55Q  but I have some issues.

I tried to connect to my Kadjar with "Car Scanner" App and it works correctly (read motor variables perfectly) and after, I tried to connect to my Zoe ZE50 with CanZE but impossible to connect it and read values.

Pop-up messages appeared: "Unrecognized ELM version response [atws<cr>?<cr><cr>>]", "Unrecognized ELM version response [atws<cr>?]", "ELM is not responding (toughness = 0)", "Timeout on [atws] []"

Smartphone information:
Samsung Galaxy S10e - Android 11

Some pictures:
[image]
[image]
[image]
[image]
[image]

I hope it is not a big issue and I am available if it is necessary to do some additional tests.

Thanks.

fesch, 2021-09-02:
Your dongle seams not to be compatible with CanZE.

https://canze.fisch.lu/elm327/

Flodesirat, 2021-09-02:
It could be possible to put a clear message in the pop-up as: "Your ODB device is not compatible" or "Your ODB device seems to be not compatible" ?

fesch, 2021-09-02:
Right, but the software is unable to distinguish the cases "incompatibale ELM" and "timeout on a compatible ELM" ...

Flodesirat, 2021-09-02:
Ok, thanks