fesch/CanZE #719: Range for ZE50

FredLeudon, 2021-09-09:
Hi all !

I just was playing with DDT4ALL and I found the ECU named CAN_MESSAGE_SET_C1A_Q4_2017_ALL_MESSAGE_LIST_OFFICIAL_01_12_2017_20180319T081912.xml which contains some datas like "VehicleAutonomyZEV" (Request 98DAF1D2).

I can't test it right now because my ZE50 is away but it may help to retrieve range for CanZE

Regards

yoh-there, 2021-09-09:
Thank you Fred! Pretty weird, as the ECU is the secure gateway!

I'll see that I can compile a _Research.csv file for the promising candidates and post it here.

yoh-there, 2021-09-09:
Unfortunately (unless I am missing something), these are not commands to the SGW, but UUDT ("free") frames, they are in the next Request Name tag :-( ID's are 0437 and 06E7, hex, names HEVC_A1 and HEVC_R11 respectively.

I don't think we have been able to get UUDT frames from the internal V bus. SIgh.

At least we know the data comes from the EVC now. That's not unexpected. Thanks again Fred.

closed 2022-01-23 by yoh-there