fesch/CanZE #721: iOS 2017 ZE40 no data

smulle48, 2021-11-14:
Hi

I've just connected to my 2017 ze40 with an kw902 Wi-Fi. All the tests are ok, but I don't get any data out of the car

Is it me that are doing anything wrong?

[image]

ik2wxx, 2021-11-16:
Hello
"NO DATA" means that OBD can't receive data from car.
Was the car powered on ? Some data can't be fetched with car not fully on.
BTW, this seems to be a bigger problem, like an incompatible version of Konnwei OBD :-(

smulle48, 2021-11-16:
> Hello
> 
> "NO DATA" means that OBD can't receive data from car.
> 
> Was the car powered on ? Some data can't be fetched with car not fully on.
> 
> BTW, this seems to be a bigger problem, like an incompatible version of Konnwei OBD :-(

If i change the car model to anything else then z40/s50 I can get data out, but how can I be sure it's correct?

ik2wxx, 2021-11-16:
Please be sure to identify correctly your car model.

smulle48, 2021-11-16:
> Please be sure to identify correctly your car model.

This is my car. So I should select ze40/ze50??

[image]

ik2wxx, 2021-11-17:
I guess you should check the car licence.
Sorry I can't help you very much in this. 
In Italy we don't have ZE40, this is a special version sold in just some countries.
Anyone can help?

yoh-there, 2021-11-17:
It's a Ph1, basically an R model with a big battery. Try to set it on R90