fesch/CanZE #722: Error messages

Rpp131, 2021-11-15:
Hi, I am a new user of Canze version 1.6 on a 2019  40kwh Zoe R110. I have the recommended Konnei code reader. When reading data, the message 'debug xxxx failed' appears frequently. Also 'Bluetooth connection lost' and 'ELM not responding....'
Typical screen shots attached.
[image]
[image]
[image]
[image]
[image]

I do collect some data but it seems slow and erratic. The Bluetooth light flashes continuously.

Do you think the code reader is faulty?

yoh-there, 2021-11-15:
There are always comms errors, and KONNWEIs are slow (but fairly reliable) indeed. My advice would be to double-check that the car tyoe is set correctly in the settings, and set the log level to None. The latter supress these popups that, by nature stay on the screen for a fixed amount of time.

Rpp131, 2021-11-16:
Thanks for the quick response.  I will set the log level to none and double
check the correct car is selected.  Is it normal for the blue tooth led to
flash continuously on the Konnwei?  The manual is not much help.

Regards

yoh-there, 2021-11-16:
It is, we're polling the car as fast as the KONNWEI will go.

Rpp131, 2021-11-16:
Once again, thanks for the quick response!

closed 2022-01-01 by yoh-there