fesch/CanZE #723: No charging history available

BuPaul32, 2021-11-16:
Version 1.60 (release-master) Build:2021-02-04
ZOE Ph2 R135
KONNWEI KW902

No charging history is shown (last 10 sessions):

Odometer	End	Type	SOC	Temp	Duration
0		---	Slow	0	0	0
0		---	Slow	0	0	0
0		---	Slow	0	0	0
0		---	Slow	0	0	0
0		---	Slow	0	0	0
0		---	Slow	0	0	0
0		---	Slow	0	0	0
0		---	Slow	0	0	0
0		---	Slow	0	0	0
0		---	Slow	0	0	0

For the rest

Charging summary
.Total battery usage (kWh)			625
.Total energy from regenerative braking (kWh)	167
.Full charges (#)				  2
.Partial charges (#)				 48

values are available

FulminatingGhost, 2021-11-16:
Hi, can you please check the Firmware Version of the EVC-HCM-VCM?
If I remember correctly, the values for history are only available upwards from a certain version.

BuPaul32, 2021-11-16:
Hi
Yes it seems that I have older versions

Diag version	07
Software	        0A10
Version		6A00

I have seen that already issue #675 reported my issue. I will try to update.

Thank you for your quick support!

closed 2021-11-21 by FulminatingGhost