fesch/CanZE #725: Unstable Bluetooth-connection with Konnwei KW903

Icoti, 2021-11-21:
Hello everybody,

first of all a description of my system:
Renault Zoe Intens R135 Z.E. 50, first registration November 2020, built October 2020. About 10.000 km.
EVC-HCM-VCM:
diagVersion:08
soft::0A10
version::69B0

OBD II adapter Konnwei KW903 Bluetooth with ELM327:
[image]
[image]
[image]

Smartphone Samsung Galaxy S5 mini SM-G800F with Android 6.0.1

CanZE 1.60 (release-master) Build: 2021-02-04 11:12:59

I bought the KW903 because it was difficult to find the KW902, and because of the smaller size, so there is less danger to hit it with the left foot during driving.
[image]

The Konnwei KW903 Bluetooth basically works with CanZE, but the Bluetooth connection is very unstable, which makes the use of CanZE with this adapter no fun.

After pairing the adapter with android and selecting it in CanZE, there will be no connection.

Using the KW903 with the Torque-app, which is recommended by Konnwei (but doesn't support battery settings), works very well, the Bluetooth-connection is perfect.

So I first connected the adapter with Torque, then started CanZE and then closed Torque.
Then CanZE will usually be able to communicate with the adapter but not systematically.

Afterwards, the Bluetooth-connection may be stable for a few minutes, or may get lost after a few seconds.
Sometimes CanZE will manage to reconnect, sometimes not and the detour over Torque will be necessary.

Screenshot with failed Bluetooth connection:
[image]

Screenshot with working connection:
[image]

So my questions are:
Is there a workaround to improve Bluetooth stability?

If not, would it be possible to improve it from the software side?
It would be really nice to have CanZE reliably working with a compact dongle.

I would also like to report the following output of the "Test ELM327 Dongle" (when Bluetooth connection is active):
"Processing prepped ISO-TP command CLUSTER SW
Requested field does not exist. This is an error in the test suite, please report"
[image]

What does this mean?

FulminatingGhost, 2021-11-21:
Hello,
the "Test ELM327 Dongle" function will need some rework since it relies on fields that are only availabe in the old Zoe. In the meantime please try the app "ELM327 Identifier" and post the results here.
Sadly the different dongles support different sets of instructions, which makes using the ELM dongles quite challenging. My guess is that your dongle does not support all the instructions we're using or something similar.
If you're handy with electronics I strongly recommend building a [CanSee](https://gitlab.com/jeroenmeijer/cansee) dongle. Checkout the Wiki page for instructions on how to build it.

Icoti, 2021-11-22:
Thanks for the reply.
Here is the result of ELM327 identifier:
[image]

Well, it may be that not all commands are supported by the KW903, though from what I experience the main problem is the Bluetooth connection. I almost steadily get the error message "Bluetooth connection lost" and "Reconnecting Bluetooth...".
Once the connection works I could access most important battery information like SoH, SoC, cell temperatures and voltage.

Do you have an explanation for the fact that apps like "Torque" or "ELM327 Identifier" manage to establish a stable Bluetooth connection to KW903 while CanZE does not?

If at least this problem could be solved it would widen a bit the hardware compatibility of CanZE and allow users to use a more compact dongle, which definitively makes sense on the Phase 2 Zoe.

About building my own CanSee dongle: Unfortunately, I never soldered electronics components, nether do I know how to flash the firmware. The instruction in the Wiki may be useful for experienced user, I cannot do much with it.

Thanks for the hint about OBDII extension cables, that is a good idea.

FulminatingGhost, 2021-11-24:
Thanks for the information.
I currently don't own a Konnwei 903 so I can't debug the issues you're having.
We do reset the Bluetooth connection if there is a problem in communicating with the dongle, so that might be the reason for the disconnects you're seeing.

Icoti, 2021-11-24:
Thanks for the reply.
I made some more tests and observed the following:
After starting CanZE, data is acquired and shown (usually) as long as I leave the app alone.
The Bluetooth connection always drops when I open the "Settings" page in CanZE.
Afterwards, the connection stays down. Restarting the app does not change this.
But if I kill the app in Android and start CanZE anew, data is shown again.

Here is a two minute log file when doing the following:
- Inserting the dongle and switching it on
- Starting CanZE (with cleared log directory)
- Opening the "Battery" page in CanZE (CanzeActivity: onCreate (BatteryActivity)). Data is shown at once.
- Leaving the "Battery" page and opening the "Settings" page without changing anything (at about 20:05:10), soon comes the message "Bluetooth connection lost..."
- Back to the "Battery" page, no new data is shown anymore, just NaN, the message "Bluetooth connection lost..." is shown again.

[debug-2021-11-24-19-45-22.log](https://github.com/fesch/CanZE/files/7598443/debug-2021-11-24-19-45-22.log)

Can you find any hints about the failure?

Icoti, 2021-12-03:
I just received an new Konnwei KW902 which actually should work with my Android device, since it is officially supported by CanZE.
Here are two pictures of the KW902:
[image]

[image]

Here the comparison KW902 and KW903:
[image]

And here is the result of ELM327 identifier with the KW902:
[image]

Icoti, 2021-12-03:
Of course I tested the stability of the bluetooth connection with the KW902. Unfortunately, it behaves the same as the KW903 with my Smartphone Samsung Galaxy S5 mini SM-G800F with Android 6.0.1.
This device is stated to be supported by the app in the app store.
Here is the log file when doing the following (same conditions like with KW903):
- Inserting the dongle and switching it on
- Starting CanZE (with cleared log directory)
- Opening the "Battery" page in CanZE (CanzeActivity: onCreate (BatteryActivity)). Data is shown at once.
- Leaving the "Battery" page and opening the "Settings" page without changing anything (at about 17:32:05), soon comes the pop up message "Bluetooth connection lost..."
- Back to the "Battery" page, no new data is shown anymore, just NaN, the message "Bluetooth connection lost..." is shown again.
[debug-2021-12-03-17-31-40 KW902.log](https://github.com/fesch/CanZE/files/7651439/debug-2021-12-03-17-31-40.KW902.log)

Icoti, 2021-12-03:
I tried to analyse the log file by looking at the lines containing "BT" or "bluetooth" and observed that at the beginning, there are some unsuccesfull connection trials until the connection eventually works.

After going to the settings page and leaving it, there are also unsuccessfull connection trials, then the app is stuck in following loop, but never manage to establish a connection:

20211203173226453,BluetoothManager: Get remote device: 0C:FC:85:17:EE:B3
20211203173226457,BluetoothManager: Create new socket
20211203173226461,BluetoothManager: Could not create RFComm Connection
20211203173226477,BluetoothManager: Cancel discovery
20211203173226487,BluetoothManager: Connect the socket
20211203173226580,BluetoothManager: Something went wrong
20211203173226585,BluetoothManager: Closing socket again ...
20211203173226588,BluetoothManager: -7 tries left
20211203173226591,BluetoothManager: Starting new try

The socket does somehow not manage to connect.
Do some of the experts around here have a clue why it is not working?

I also wondered why the message "Could not create RFComm Connection" comes after "Create new socket" and had a look in the source code at following code file:
app\src\main\java\lu\fisch\canze\bluetooth\BluetoothManager.java
At line 137, a "d" seems to be missing in ...(secure ? "createRfcommSocketToServiceRecor" : ...

But I think this is not the primary cause of the unstable bluetooth connection.

Icoti, 2021-12-03:
I also performed trials with another, more recent Android device: Galaxy Tab A (2018, 10.5) SM-T590.
With this device, I could not reproduce the connection problems for KW902 as well as KW903.
Here is a log file under the same conditions as mentioned above (with KW902):
[debug-2021-12-03-19-58-06 902 Tab A.log](https://github.com/fesch/CanZE/files/7651933/debug-2021-12-03-19-58-06.902.Tab.A.log)
Maybe it helps understanding the problems with the S5 mini.

yoh-there, 2022-01-01:
We are in a bit of a pickle here. Where at least I myself hate throwing away good devices, an S5 mini is by now a 7.5 years old device and the latest Android update on it (6) is 2% of the of the user base of CanZE. Add to that there is at least a non-zero chance it is a problem with your specific device, us not having a S5 mini, and on top of it absolutely nothing we can do on the hardware and the Bluetooth managing OS level. I think you can see this would almost certainly be a disproportional exercise. Sorry I do not have better insight.

Icoti, 2022-01-02:
If no one sees the cause of the bug with low effort I understand that debugging for this old device would not be proportionate.
Apart from that, would it make sense to correct the file BluetoothManager.java as mentioned three posts above?

yoh-there, 2022-01-02:
I'll have a look at that, but there is a another tech thing lurking that we need to fix first

yoh-there, 2023-10-02:
Closing because of inactivity

closed 2023-10-02 by yoh-there