fesch/CanZE #728: Renault ZOE ZE50 DTC menu greyed out

Dan3333333, 2021-12-10:
Hey,
Can you please tell me why my DTC menu is greyed out and how can I access it.

Adapter used is KONNWEI KW902 wifi. iOS 15.1
[image]
[image]

Dan3333333, 2021-12-10:
Same for ELMT testing
[image]

ik2wxx, 2021-12-10:
ELM testing not needed as the test is in the Settings screen

Dan3333333, 2021-12-10:
Then how can I test my ESC system for example? Thought it was is ELM testing menu

Dan3333333, 2021-12-10:
Ok, thanks for clarifying this… waiting to find out how can I test my ESC system and how to read DTC codes

ik2wxx, 2021-12-10:
DTC screen is not implemented yet in iOS version.
I don't have any idea on how you can test ESC system with CanZE.

Dan3333333, 2021-12-10:
Regarding ESC see https://youtu.be/Fm5bPX5TM8c

Dan3333333, 2021-12-12:
Just installed your app on an Android phone and DTC menu is not greyed out :( but don't have a bluetooth dongle to use it

Dan3333333, 2021-12-18:
Any updates on greyed out dtc menu on iOS?

ik2wxx, 2021-12-20:
iOS development is currently in standby if no other users are willing to partecipate

closed 2021-12-20 by ik2wxx