fesch/CanZE #729: R90 no tire data

smulle48, 2021-12-10:
Hi

I don't get any tire data. This is how it looks (I know it's not connected now)

Can't I read and write tpms id on iOS?
[image]

ik2wxx, 2021-12-10:
Tires info is not fully working as it's not tested.
We can't find anyone with iOS and a suitable Zoe to continue development of this section.

smulle48, 2021-12-10:
> Tires info is not fully working as it's not tested.
> 
> We can't find anyone with iOS and a suitable Zoe to continue development of this section.

Thanks for the info 😄 

If you need help, I would be more then happy to help you out with whatever I can.

Though I have not developed on iOS before, I've been programming for XboxMediaPlayer (Kodi) among other.

Does the tire page work on android?

yoh-there, 2021-12-10:
It does, at least when I made it long ago. There is a slight chance something has changed with an updated BCM / BCM firmware.

siavashvj, 2021-12-12:
Similar issue for R110. Never ending loading of data files / assets. The Read/Write buttons are greyed out. 

I have a KW902 + R110 2019 Zoe + iOS (WiFi). 

Uploading IMG_8394.MP4…

ik2wxx, 2021-12-13:
> If you need help, I would be more then happy to help you out with whatever I can.
> 
> Though I have not developed on iOS before, I've been programming for XboxMediaPlayer (Kodi) among other.

The sources are online, if you want to collaborate.

closed 2022-05-31 by yoh-there