fesch/CanZE #732: CanZe and Renault Zoe R135 ?

JustinShel, 2021-12-21:
Should App
v: 1.60 Build 2021-02-4 on Android and Konnwei bluetooth dongle work with Zoe R135 

Getting Debug, followed by number and fail at bottom of most screens ?

yoh-there, 2021-12-21:
Should work. Make sure you have carefully gone through settings and have looked at https://canze.fisch.lu/qa/, esp the 9th question.

Please close this issues if solved, thanks.

JustinShel, 2021-12-22:
> Should work. Make sure you have carefully gone through settings and have looked at https://canze.fisch.lu/qa/, esp the 9th question.
> 
> Please close this issues if solved, thanks.

I have successfully paired the Konnwei with my Pixel 4a android phone - I can see it on the settings screen. I see a couple of German users have successfully used the same device and canZe to connect to R135 according to their Amazon feedback (translated)[ ](https://www.amazon.co.uk/dp/B00W30RMEC#customerReviews)

yoh-there, 2021-12-22:
If you have selected the proper paired dongle AND the proper car (ZE50 for an R135), it should work. If it doesn't I suspect a dongle problem unfortunately.

JustinShel, 2021-12-22:
> If you have selected the proper paired dongle AND the proper car (ZE50 for an R135), it should work. If it doesn't I suspect a dongle problem unfortunately.

Ahh ok, think I tried R110 rather than ZE50. Will let you know

yoh-there, 2022-01-01:
Any success? If so please close the issue. Thank :-)

JustinShel, 2022-01-05:
Set to ZOE ZE40/ZE50 and working fine!

Thank you for your assistance

closed 2022-01-05 by JustinShel