fesch/CanZE #736: KONNWEI KW902 bluetooth 3.0 not connecting

electricstrm, 2022-01-07:
Hi, i bought this dongle from amazon uk. The advert says bluetooth 2.1 but when i got it, it was a more recent version with bluetooth 3.0.  I have late 2018 Zoe R110 ZE40.  I plug the dongle in, press power. The buttons flash.  I tried connecting 2 different phones. First my pixel 3a and a Samsumg S10X..  Both uses bluetooth 5.0, both fail the same way.  The phones see the dongle and it's address. I get asked a password. The dongle manual says to enter 1234. So i do that. It pairs, but then never connects.  I tried connecting from the bluetooth menu on the phone and also by going inside the Canze app.  Is the dongle broken ? or is it a compatibility issue ?

https://www.amazon.co.uk/Konnwei-Kw902-Black-Bluetooth-2-1-OBDII-Diagnostic/dp/B00W30RMEC/ref=sr_1_5?crid=3FPVQAD5TWDBH&keywords=konnwei+kw902&qid=1641560953&sprefix=%2Caps%2C278&sr=8-5

Btw i speak english, french, italian and spanish if you want to answer in another language. Also i'm a software developer.

electricstrm, 2022-01-07:
The small manual only talks about the torque software, no mention of ELM. The box says ti supports all ODBII protocals, has a fast ARM chip. Supports J1850 VPW, J1850 PWM, ISO9141-2, KWP2000, CAN

fesch, 2022-01-07:
"fast ARM chip" does not sound very well :-/

If your also a developer, you will certainly not have big problem in trying this: https://canze.fisch.lu/dongle-test-script/

KarelSvo, 2022-01-07:
https://cvtz50.info/en/elm327/

electricstrm, 2022-01-08:
> https://cvtz50.info/en/elm327/ https://play.google.com/store/apps/details?id=com.applagapp.elm327identifier&hl=en&gl=US

those are really helpful.

electricstrm, 2022-01-08:
> "fast ARM chip" does not sound very well :-/
> 
> If your also a developer, you will certainly not have big problem in trying this: https://canze.fisch.lu/dongle-test-script/

Unfortunetly my phone just wont even connect to the dongle. I used developer mode on my pixel to change bluetooth version and still no success. It pairs, but wont connect. Also i asked on amazon what version of elm is supports and they told me it supports Nissan and Leaf ? 

I then did some searches on ebay, the sellers have a model checker.. and even dongles that advertised supporting Renault dint support my car !! like this one
https://www.ebay.co.uk/itm/193728676912?fits=Car+Make%3ARenault&epid=10020711361&_trkparms=ispr%3D1&hash=item2d1b210430:g:RZoAAOSw6D1bKwVu&amdata=enc%3AAQAGAAACoPYe5NmHp%252B2JMhMi7yxGiTJkPrKr5t53CooMSQt2orsSd3M8ngECaRA0SmyU4KT%252Fni2JvQAL8gQ2Qwd1%252BQlW8PHG7RS%252B1N%252FSpZklMFzNoAOfEr%252B5MrSWz8WT%252BmsMYyGqtp4mSXJ5IA9V1MIK%252BYX55Un9dG25GsW%252F9eBIGuCG5V%252ByD8vg2G61djCW9PQfI5mSitKG%252F9XndjT481Z9Kd%252FGT2g9E25G%252BftWJVY5SPwigRon0%252BefV%252FeoHzIWsyFEUs9Yl5HL75pfaUOxZMPw3HmU2SdxYim%252FA0aBkfkAP0NmDkj4TKHdyyXkyw5JAXSsKb9Ur8%252BCq5AOF6KrWdsYni83xmkf%252B5gEunQkU%252B3lI2oupRD2xMHjKmrblFUjzYvASKiWap%252B46R5NYLxAWnMRcq%252BCurXLpkm5TOH4d2djH1I3ilqtJ3eT5P8kTZe8KstBxfp48hwOJ3Ba%252Bt7dHSDjouzPpnKILRwmR8GV4VCajSEh0uzph11L7RotRrZrIydJ4Xkg7iaWutPWLYl3Zk2MQ2iyU4mMoYRNbqKnFZdaeYF%252FSQdof7KVWzrXJJThL1SMlOv40%252FqbsmU%252FKisj3wQn9Kd%252B4CH80jqKiz7Df3E%252BDV0FMRneQO%252B3yR2zmJvEdkZQf69yImlYQ3G%252BK%252FqKieLV%252FUtlfeZAI1i3%252FULczoJGyq6FSkwOpvEUHgYdI8acTVkOSb4w2BEv7WpEhoaBYoEQxGzpdOI3nuHuWHXWl7Jk0CLlfGUvbqk%252F86%252FJ5fTV1qQM33bXQcShyu%252BWxsf2wqe7hX4g7YNkaoXBMaXGGcvkTOdINy991kEBTRbXH%252FzIMYFrEXcSKJwOWamz5JkcL3Rz9Kwm9F1jtkjIOYtN%252F7RoVesdM7x%252BntNEdp2TPeFzEv9QnJtgHw%253D%253D%7Cclp%3A2334524%7Ctkp%3ABFBMotHcnsdf

electricstrm, 2022-01-08:
I have about 2 options left.. i'll let you know how it goes

rangleri, 2022-05-25:
_> Hi, i bought this dongle from amazon uk. The advert says bluetooth 2.1 but when i got it, it was a more recent version with bluetooth 3.0. I have late 2018 Zoe R110 ZE40. I plug the dongle in, press power. The buttons flash. I tried connecting 2 different phones. First my pixel 3a and a Samsumg S10X.. Both uses bluetooth 5.0, both fail the same way. The phones see the dongle and it's address. I get asked a password. The dongle manual says to enter 1234. So i do that. It pairs, but then never connects. I tried connecting from the bluetooth menu on the phone and also by going inside the Canze app. Is the dongle broken ? or is it a compatibility issue ?_

Hello! I was struggling with this same issue, you described. The dongle is Konnwei KW902, but BT 3.0 version(It says OBDING BT 3.0 on it). The package also says it is using the fastest ARM Chips etc.
Android version on the phone used was 11(motorola 5G plus). And the car Zoe R110 z.e 40. Bluetooth device could be paired, but there was no connection when using the CanZE app, BT symbol stayed grey and nothing worked. 
But when I tried with an android 7.0 phone (older Motorola moto C plus), the connection works, BT symbol in the CanZE app becomes active and connection works. I was able to read a fair amount of data from the car but not TPMS sensors for example.

rangleri, 2022-05-31:
An update to the previous post, this particular dongle started working also with the newer Android device. I changed the BT device in the CanZE App settings momentarily to an another paired BT device, then selected KONNWEI again, and it worked! BT symbol became blue and active. Also the TPMS data could be read and written. It is a fantastic app!

yoh-there, 2022-05-31:
@rangleri  Thank you for updating the issue. Enjoy the app.

@electricstrm hope you found a working dongle.

usernamepasswordBS, 2023-03-05:
Sorry if I should start a new thread but my similar issue is this:

I bought the Konnwei dongle a year ago, it worked fine with Canze although i didn't explore much, just the battery SOH.
Anyway, the other day I had a fault and tried to use Canze again but no bluetooth connection, keeps saying "reconnecting bluetooth" when i press the connect button but doesn't connect.
I tried the Konnwei plus another obd app and that works fine so I believe the Konnwei and the vehicle are both OK.

I tried an old phone which definitely worked with Canze before, but no good now.

What should I try next?

yoh-there, 2023-10-06:
Closing (stale)

closed 2023-10-06 by yoh-there