fesch/CanZE #737: Android Auto projection

iboboc, 2022-01-23:
It will be a great feature if CanZE can have a project screen for Android Auto. 
If the electric car has support for Android Auto, instead of viewing all the car data on the phone screen, it will be available on the car screen.
The change will not affect the current functionality, no visible changes for current application use, but once connected to headunit, the Android Auto view can be enabled.

The implementation is very simple, a nice article is here https://medium.com/@dirkvranckaert/my-journey-on-developing-a-custom-android-auto-app-de4fb7777fd1 and sample code is available in github https://github.com/dirkvranckaert/AndroidAutoHelloWorld

In a first step, only basic info can be displayed and then, full functionality can be added.

I am attaching a sample screen with the most important functions.

[image]

Thanks.

yoh-there, 2022-01-23:
Thank you for the suggestions and the linked article.

The article outlines fairly precisely why it probably won't happen though: this can't be distributed through Google Play but after serious investment to get the green-light from Google. A non-Google Play distribution is out of the question but for the geeky users.

Add to that that the Ph1 can at best be coaxed into running Android Auto and is super slow when you do. I myself own a Ph1 car.

Note that this is my personal opinion. If anyone in or out of the current developers is happy to try this and add it as a flavor to the repository such that a non-AA compilation remains possible without too much hassle, I don't see why we couldn't incorporate it.

iboboc, 2022-01-23:
Thanks for feedback. The article is old (2017) when indeed was not possible (only in developer mode), it was just for reference.
But now it is. Recently, Google approved and it is possible to publish it. For eg. Plugshare it is published and working fine, see source here: https://github.com/PlugShare
At least, is what I know. Thanks.

yoh-there, 2022-01-23:
Cool, and missed the 2017. That should make the AA app base a lot easier / larger.

Just to make sure there is no misunderstanding: I myself have no intention of building this. That is no dismissal of the suggestion, just other priorities in life, and owning this Ph1 car. But as I understand it the hurdles to get it done in general are lower than it was before, which is good.

iboboc, 2022-01-23:
Understood, actually there is no expectation from my side for someone to pickup this... is purely voluntary work, I understand that.
It is just a nice-to-have highly appreciated feature, at least from my side.

KarelSvo, 2022-01-26:
https://youtu.be/NJTe-RcivGo?t=1325
https://www.youtube.com/watch?v=-mEeY4vMLa8

iboboc, 2022-01-29:
I started myself a trial. Forked the project, added the AA support in the manifest and created the activity and service, change de sdk to 31, but I have a build error which I cant understand.

The error is 

A failure occurred while executing com.android.build.gradle.internal.res.ResourceCompilerRunnable
> Resource compilation failed. Check logs for details.
  > Failed to compile values file.
  
 Looking in detail, 

:app:mergeFdroidDebugResources FAILED
/home/user/StudioProjects/CanZE/app/src/main/res/values/attr.xml:2:4: Duplicate value for resource 'attr/backgroundColor' with config 'DEFAULT' and product ''. Resource was previously defined here: /home/user/.gradle/caches/transforms-3/ce0f46f6f820154d926e53f23e6638ad/transformed/material-1.4.0/res/values/values.xml:2976:4: .

Which doesn't say much, so I'm stuck.
Any help, appreciated.

iboboc, 2022-02-06:
Since nobody was picking up this feature request, I've started learning Android myself :).
After an initial fork of the master branch, I've successfully build the original source with minor changes (update gradle, sdk, etc).

The next step was to add support for Android auto, namely changes in manifest file and created two new files, service and screen for Android Auto. The screen is a grid with 6 items which shows 6 important data from OBD. Currently values are dummy (hard encoded), to be added/updated in a later commit.

The test was completely successfully, in order to test it, I've used the Google tool DHU Desktop Head Unit 
(set mobile to development mode and Android Auto to develoment mode (tap 10 times on version), plug in your mobile to your development computer with a USB cable, start the Android Auto head unit server, start the desktop head unit: 

<SDK_LOCATION>\platform-tools\adb forward tcp:5277 tcp:5277
<SDK_LOCATION>\extras\google\auto\desktop-head-unit.exe

https://developer.android.com/training/cars/testing

This change does not affect the application functionality in any way, it will be visible only if you are using Android Auto.
Not sure if it is useful to anyone, it is to me, but it is a nice feature to have at least from the development point of view.

Unfortunately, in order to be visible and working in the car, it needs to be merged and publish to Google App Store (at least beta channel), going through Google review: https://developer.android.com/training/cars/distribute

[image]

iboboc, 2022-02-15:
First draft published to Play Store in order to test it in car.

Will continue with adding support for Spring before sharing the AA changes.

[image]
.

romelec, 2023-02-17:
@iboboc did you continue to work on it ? I noticed no commit on your fork.
I never made android apps but could try to continue the dev.

iboboc, 2023-02-18:
Hi @romelec ,

While the AA implementation is working, I have issues to publish it widely available to anyone in Play Store, since Google has very strict rules regarding the use of an application in the car (defining the right category, permissions, usage, etc), so I am keep working to make it compliant. Once I'll have some success, I'll let you know.

yoh-there, 2023-10-02:
Closing (stale). If there is any development we can pick up / integrate, we'd be more than happy to!

closed 2023-10-02 by yoh-there