fesch/CanZE #742: Suport for Dacia Spring

iboboc, 2022-02-25:
Dacia Spring shares the same structure of Zoe, so with some changes, it can be supported as well, knowing that already are thousands of owners.

I intend to make those changes, if there are developers that wants to help, let me know.
In the same time I need to know if there is interest from the owners of the code, willingness to merge those changes and support Dacia Spring as well.

yoh-there, 2022-02-28:
As far as I am concerned (I am not the only one here in town!) there is absolutely willingness and enthusiasm to include the Dacia Spring, or any comparable car really. Really the only thing is the original authors do not own such cars and we had a hard time getting ZE50 to work properly. Basically it was only resolved when committed developer-owner were on board. Seems like you are in that position and that is wonderful!

Personally I have no problem merging in PR's especially into beta, and to put some announcements out, as long as it doesn't break current functionality of course.

Calling on @fesch re. his opinion (and the others chime in too of course!)

Leobaillard, 2022-02-28:
I'm with @yoh-there on this one. We suspected that the architecture would be similar, allowing for a common base.

I only have a ZE50 but I'm willing to assist your efforts in any way I can. Feel free to ask for help if need be. Maybe we can share a more direct communication channel than the Github issue if you need extended help.

fesch, 2022-02-28:
Yes, as long as the base of CanZE stays, the same, merging will be fine I think ...

yoh-there, 2022-05-31:
Thank you for all your fantastic work @iboboc We'll try to roll up your PR soon and then announce on the blog.

closed 2022-05-31 by yoh-there