fesch/CanZE #744: Canze doesn't do anything keeps hanging on debug

bruvv, 2022-03-11:
When using the icar pro sometimes it works just fine and I can read out everything and sometimes it just doesn't do anything and when I connect to the dongle using canze on the bottom it shows: debug debug# and doesn't get any data. I test the dongle and it sort of looks good I guess ?
[image]
[image]

bruvv, 2022-03-14:
Can this be related to my other issue https://github.com/fesch/CanZE/issues/743 ?

bruvv, 2022-03-21:
Hello. Anyone?

bruvv, 2022-03-21:
Did you had a change yet to help me out @yoh-there @fesch ?

fesch, 2022-03-22:
I have no idea about your problem, but it it know that ELM's are not amongst the reliablest beasts :-/

bruvv, 2022-03-26:
Thanks but can you see anything in the log ? It's also weird that in one car it works better, q210 then in mine r240

yoh-there, 2022-05-31:
I would suggest to try the KONNWEI 902. I understand it is frustrating when having an (almost) working setup, but some dongles just don't work in all situations. Closing as the issue is stale.