fesch/CanZE #745: Problem with Android car radios

a68k, 2022-03-12:
Android radios have a huge problem pairing with OBD2 dongels.

I was converting in my Zoe from R-Link to an Android Radio now.

The first radio, it was hard to find a OBD2 dongel that pairs with the radio (whatever the android developers screwed up here...) and that can communicate with the Zoe's can bus. I had a OBD2 dongel from a friend, that worked, searched for weeks, found a used one, 
That worked so far in most times.
But the fiorst radio had some issues and I had to return it. So I bought another one, another model. I didn't think, that similar looking radios (for sure from the same maker) can be so different...
The second radio was a nightmare in many ways, so I returned it too and now I have the third radio in the car. 
I am tired of again send back and order another radio.

The third radio can pair with with these two dongels I have.:
This one : https://i0.wp.com/canze.fisch.lu/wp-content/uploads/2015/11/elm-no-5.jpg
and this one: https://www.computerworld.ch/img/7/4/8/4/3/7/1055708_w400_h284.jpg

But can't pair with the KONNWEI or iCarPro I use with my phone in Zoe.
(KONWEI it shows a popup [0000] but that's not the key)

But all dongels work with my Android phones and laptop.

CanZE let me select the OBD-II displayed dongel.
The blue BT animation shows, it is communicating, but the CanZE is not getting datas.
After 1,5 minutes in one case when I was filming it weeks ago (I did a video series in YT) it was working good then.
At the end I hoped that's how it at least will work, better then it is not working.
But in the next event, it was not working again. Waiting minutes and minutes.
So untill today,

On the ELM327 Test page in CanZE it tells me:
to wait untill the background process is ended. No connection. Close display and check connection

But WHAT background process? How to find out and end it / make sure, it will not block it again?

On the radio is Android 11 (at least what it shows me...)

Can USB cable be added to CanZE? Will this work?

Thanks!

iboboc, 2022-03-12:
What Android radio is that?

NLFiscaal, 2022-03-12:
We have no facilities build in for a cable connection. To be honest, the need hasn't come up and there would be the whole grumbling about not having OTS on the phone etc etc.

Can I make an out-of-the-box suggestion? Build a CanSee dongle. It won't be 20 Euro and a bit of soldering (you don't seem shy on taking on a serious project!). It is way way faster, and way more stable, partly because it doesn't carry all the ELM overhead, partly because it does the entire ISOTP protocol, thereby reducing the turn around times between dongle and phone massively.

fesch, 2022-03-12:
And CanSee dongle is ways faster than any ELM327 based one ;)

a68k, 2022-03-12:
out-of-the-box suggestion are most the best ;)

I will search for it, thanks

Am 12. März 2022 13:40:43 MEZ schrieb VictorMaes ***@***.***>:
> We have no facilities build in for a cable connection. To be honest,
> the need hasn't come up and there would be the whole grumbling about
> not having OTS on the phone etc etc.
>
> Can I make an [...]

a68k, 2022-03-12:
So ok I found the right archive to download.
https://gitlab.com/jeroenmeijer/cansee/-/archive/master/cansee-master.zip
Will it be able to pair with the Android radio via BT?

Is this the correct ESP?
https://www.amazon.de/ESP-32S-ESP32-Development-Bluetooth-Antennenmodul/dp/B071JR9WS9
or
https://www.amazon.de/SP-Cow-Development-Mikrocontroller-Antennenmodul-ESP-WROOM-32/dp/B07VJ34N2Q
https://www.amazon.de/AZDelivery-NodeMCU-Development-Nachfolgermodell-ESP8266/dp/B074RGW2VQ

(don't want to wait untill June for shipping from CN to Germany)

I have one or two of these at home
https://www.amazon.de/SN65HVD230-Connecting-Protection-Communication-Development/dp/B09GF968MH

but the linked ones are smaller, so these are ok?
https://www.amazon.de/ICQUANZX-SN65HVD230-CAN-Bus-Modul-Kommunikationsmodul-CAN-Bus-Transceiver/dp/B07VG3Z9GT/ref=sr_1_2?keywords=SN65HVD230&qid=1647094752&sr=8-2
https://www.amazon.de/ICQUANZX-SN65HVD230-CAN-Bus-Modul-Kommunikationsmodul-CAN-Bus-Transceiver/dp/B07VG3Z9GT/ref=sr_1_2?keywords=SN65HVD230&qid=1647094752&sr=8-2

Two can bus decorders = it can communicate with the main CAN and the "multimedia" can? great! ;)

All other things I have (old / broken OBD2 adapters and plugs, I can take the connector from)

But someone will have to give me some help, how to program the ESP then.
I can't find the documentaion for it, showing the steps :(

I did programmings before, but it is happening once a year or all 2 years, so ... I am not trained in this.
But I could make a "how to" in german in addition to my Android Radio Project, so other can benefit from it then.

NLFiscaal, 2022-03-12:
Modules you have should be OK, though our experience with the 3.3 volt units have been mediocre at best, mostly because of fake chips. Nowadays, we advise to use the 5v TJA's power the boards with 5V and let the regulator on the ESP board create it's own 3.3 regulated. It is wise to add a small voltage divider in the line from the TJA to the ESP, as the EPP32 is not 5V resilient on it's input. But if yours work, all the better.

We never finished proper dual bus, so while the hardware can do it, the firmware doesn't support it. As the new ZOEs only have one bus to the OBD2 connector, please consider this an abandoned idea.n The ESP32 has one internal CANbus controller anyway, so the only way to do this is switch back and forth. Very mêh.

The power you would have to rewire: as per OBD standards, the connector is always powered on.

Programming: it's fairly straightforward. You don't need to do any programming, just flash the firmware on it. iow, you need to feel comfortable with the toolchain. I prefer VSCode with the platformIO extension. That will more or less do all the heavy lifting for you. We could give you a binary, but getting the flashing software up end running will be roughly the same pain.

Hope this helps.

yoh-there, 2022-03-12:
Modules you have should be OK, though our experience with the 3.3 volt units have been mediocre at best, mostly because of fake chips. Nowadays, we advise to use the 5v TJA's, power the boards with 5V and let the regulator on the ESP board create it's own 3.3 regulated. It is wise to add a small voltage divider in the line from the TJA to the ESP, as the EPP32 is not 5V resilient on it's input. But if yours work, all the better.

We never finished proper dual bus, so while the hardware can do it, the firmware doesn't support it. As the new ZOEs only have one bus to the OBD2 connector, please consider this an abandoned idea.n The ESP32 has one internal CANbus controller anyway, so the only way to do this is switch back and forth. Very mêh.

The power you would have to rewire: as per OBD standards, the connector is always powered on.

Programming: it's fairly straightforward. You don't need to do any programming, just flash the firmware on it. iow, you need to feel comfortable with the toolchain. I prefer VSCode with the platformIO extension. That will more or less do all the heavy lifting for you. We could give you a binary, but getting the flashing software up end running will be roughly the same pain.

Hope this helps.

yoh-there, 2022-03-12:
Apologies, I was logged on as a different user. Here is my original comment to your post.......

We have no facilities build in for a cable connection. To be honest, the need hasn't come up and there would be the whole grumbling about not having OTS on the phone etc etc.

Can I make an out-of-the-box suggestion? Build a CanSee dongle. It won't be 20 Euro and a bit of soldering (you don't seem shy on taking on a serious project!). It is way way faster, and way more stable, partly because it doesn't carry all the ELM overhead, partly because it does the entire ISOTP protocol, thereby reducing the turn around times between dongle and phone massively.

a68k, 2022-03-12:
ok thanks, I will check the hardware I have at home (I am still at work, cinema...) 
may be I have everything by hand to build it. The bigger PCB should also work fine, can remove the screw terminals

ok, only main CAN, but that's ok. Would have been a really great thing ;)

Another idea I had, the OVMS 3.x also has the ESP ... the BT is still unused ... 
But there is no one programming the firmware, so it could also act as a BT OBD2 device ;-)

Programming: "VSCode with the platformIO extension" I remember from another project, that was not continued and I wanted to build this, should be still installed on my PC

I check this

a68k, 2022-03-12:
ok .... yes I had one in the box, also the can pcb (with screw terminals, chip is a VP230)

I installed the VSC, also Python3.5 (as it is a W7x64 machine), found a "getting started with..." https://randomnerdtutorials.com/vs-code-platformio-ide-esp32-esp8266-arduino/#2 and ... opened the folder of the unpacked archive, ... tried ... ;)
Clicked on the Alien face, "upload" and it started compiling (?)  so it seem it worked.

If that was all, too easy ;-)

`...
Wrote 1365264 bytes (821215 compressed) at 0x00010000 in 21.7 seconds (effective 504.1 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...`

Now let's wire it up and test it later the day.

Do I need the switch on the ESP?

[image]

a68k, 2022-03-12:
oups it has different pinout
https://learn.adafruit.com/assets/41624

but if I guess correct, some match:

VIN = USB?
D27 = 27/A10
D26 = ?
D25 = ?
D33 = 33/A9
D32 = 32/A7

D4 = A5/4
D5 = SCK/5

?

Will be better I order the ESPs from Amazon.

a68k, 2022-03-14:
woohoo, that was fast ... Amazon ordered ESPs just arrived

Here the log of programming. Is this all ok?
[log.txt](https://github.com/fesch/CanZE/files/8245120/log.txt)

Will assemble the dongle now...

yoh-there, 2022-03-14:
Log looks good to me

a68k, 2022-03-14:
thanks

I have now also changed the line "cs_config.mode_leds" form "single led" (1) to "multi LED" (2)
Just searching for a blue LED haha ... what is the function of each LED?

Blue will sure be for active bt connection?

yoh-there, 2022-03-14:
#define LED_BLUE 25   // BT connected, connected to WiFi, client has connected
#define LED_GREEN 33  // CAN traffic, either sending or receiving
#define LED_RED 32    // power indicator
#define LED_YELLOW 27 // reserved for future use (maybe: off = BT mode, on = WiFi Client, blink = WiFi AP)
#define LED_WHITE 26  // reserved for future use

a68k, 2022-03-14:
ok ... done ;) 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

On my Android phone it works.

But on the Android radio it doesn't work.

This first test was done with unchanged options in the config file(s)
LEDs didn't work, I check this after uploading the changed config file to the device.

It found the device in the first scan and paired it? But it didn't find / display it anymore when searching... mystery.
But in another app it was found and displayed:

Video when the other app found it and CanZE can't get datas: https://youtu.be/tIvTbaW8sQc
I also tried "isotp fields" (I am not exactly sure what it does), but also don't work ...

ok - so I checked in CanZE, it is displayed there and I can select it. Great.
But CanZE gets no datas to display, like it is with the other dongles.

There is something interrupting in the radio?

I don't understand this .... (I can imagine it all on my way, but in deep topic I am an "idiot")

And you get no usefull answers from the Chinese sellers. 
The buy that all in a big lot and sell it.
It is a lottery win, if you get any reply from them. So useless it in most cases is.

I am sure now I am at the wrong place ... but I ask it ;)
How can I find out what is blocking / interrupting this all...?

The radio pairs with my phone and that works like it should.
Why can't the radios also work like an Android device...

a68k, 2022-03-14:
Just meassured on the pins for the LEDs , just some 100-500 mV on the pins.
Do I have to set something more then telling him "Multi LED" ?

a68k, 2022-03-15:
someone?

yoh-there, 2022-03-15:
I don't think anything else but multi (and of course #defining/using the proper pins, polarity, etc etc) is needed.

I have no explanation for the issue with the Android radio, sorry

a68k, 2022-03-15:
ok, then this is mysterious ... 

I just made some photos... maybe someone knows what to do with it.

Information about the Radio BT ("Bluetooth Scanner")
[image]

Output information about the CanSee adapter ("Bluetooth Scanner"):
[image]

Then in the developer options, maybe there is something else? Doesn't look like anything to me to understand what it means
It is in german ... sorry

[image]
Betriebszeit = running time

[image]
- Show BT devices w/o names (shows only mac address then)
- deactivate absolute volume (max. volume if haveing problems)

[image]
- Abtastrate = sample rate
- Systemauswahl = pre selected by system (default?)
- Audiokanal = audio channel
- Modus = mode
- verwenden = to use
- Wiedergabequalität = quality of playback

[image]
- Maximal Zahl verbundener BT Audiogeräte = maximum number of BT audio devices

yoh-there, 2022-03-31:
Great, thanks for reporting back. Seems a specific thing with the auto-device somehow and since CanSee works and we have no way to change the cheap dongles, I am closing this issue.

closed 2022-03-31 by yoh-there