fesch/CanZE #746: HTTP-Connections

littleyoda, 2022-03-27:
In both the Android and IOS versions, there is an option to choose an HTTP connection instead of a dongle.

I'm playing around with the HTTP-Gateways right now. 
But with varying success.

In the Android-Version I could not get the function to run at all.
(Settings: HTTP Gateway, 192.168.2.150, Http).
In the IOS I was able to see requests like "GET /?command=i18daf1da,1003,5003 HTTP/1.1"

1. Are the sources of the HTTP "Backend" that you are using for testing available?
2. Ideas why the IOS version works and the Android version does not?

Thanks

yoh-there, 2022-03-27:
Http is indeed a function for testing / development purposes. With some (limited) luck, it reduces the constant back and forth with laptop and phone on cold winter nights ;-)

There is PHP source at least statically emulating some of the Ph1 model. It's extremely ugly code though (basically a huge case statement). If you can setup a php host then I could probably share the code on gitlab. Please remind me if I forget, busy day ahead.

yoh-there, 2022-03-28:
Silly me, was already done. Here is the link

https://gitlab.com/jeroenmeijer/canseeemulator

Closing this issues as resolved.

closed 2022-03-28 by yoh-there