fesch/CanZE #747: CanZE/blob/master/app/src/main/assets/ZOE_Ph2

zoetime, 2022-04-01:
Hello developers
I want to understand how your software works. 
For example this line  https://github.com/fesch/CanZE/blob/master/app/src/main/assets/ZOE_Ph2/_Fields.csv#L38
In the ddt2000 RBMS_MCPU file is written Step=" .0625" Offset="-40"  and you write in CanZE,.0625,640, 
Which formula do I need to convert the offset value from ddt2000 so that it fits CanZE? 
or this line  https://github.com/fesch/CanZE/blob/master/app/src/main/assets/ZOE_Ph2/_Fields.csv#L16
ddt2000 says Offset="-32767" and you write in CanZE ,1,32767,  Here only the sign changed.  I don't understand according to which rules the offset is converted to CanZE. I would like to understand the system. 
Thx

FulminatingGhost, 2022-04-01:
Hello,

The offset is applied prior to scaling in CanZE ( 0.0625 * 640 = 40) and I think the default behavious subtracts the offset, hence the sign change. However we did some extensive testing and found some offsets in the original files to be incorrect. So you might find values that don't correspond to the values found in the DDT2000 files.

zoetime, 2022-04-02:
Hello [FulminatingGhost](https://github.com/FulminatingGhost)
Thank you for your quick response and explanation. 
So the offset must always be inverted. If ddt2000 "Step" is present, the ddt2000 offset must be divided by "Step". "Step"  equals  CanZE resolution.
If ddt2000 "Divide By" is present, then 1 divided by "Divide By" equals CanZE resolution. In this case, the ddt2000 offset value is accepted and only inverted? 
But I don't understand how, for example, this line was converted  https://github.com/fesch/CanZE/blob/master/app/src/main/assets/ZOE/EVC_Fields.csv#L832
DivideBy=" 13107200" is written for this value in the ddt2000 EVC_3180 file.

FulminatingGhost, 2022-04-02:
It is always represented as a factor (1 / 13107200 = 7,62939453125 * 10^-8).

zoetime, 2022-04-02:
Would ,.000000076, work too?

yoh-there, 2022-05-31:
Yes

closed 2022-05-31 by yoh-there