fesch/CanZE #748: CanZE esp dongle as MQTT client?

smulle48, 2022-04-03:
Hi

I was wondering if anyone have tried to setup the CanZE esp dongle as an mqtt client?

The reason is that the provider of my charger unfortunately don't give access to the api of the charger, so I thought it might be possible to modify the code for the CanZE dongle so it (when plugged in) connected to my home wifi and started pushing data to my local mqtt server?

This way I could avoid modifying my breaker box :-)

yoh-there, 2022-04-03:
The CanSee dongle is a passive device. It doesn't do ANYTHING spontaneously. It sure could be programmed that way, but it isn't and at least from my end not planned. However, it can be contacted over TCP/IP socket over WiFi. It is very simple do do what you want with ie NodeRed. That would separate the somewhat delicate CANbus codebase from whatever fun things you want to do.

closed 2022-05-31 by yoh-there

SMCinc, 2025-07-29:
I use my CANSee dongle in WIFi mode and have a simple bash script in a PI called every minute by crontab. 

` sh #! /bin/sh

rd=/ramdisk/obis

plug=$(tail -n 1 "$rd/plug.txt") 

if [ $plug -gt 1 ]; then

soc=`echo "i7ec,222002" | nc 192.168.2.81 35000 -w 1 | cut -c 11-14 | tr '[:lower:]' '[:upper:]' ` &&
soc=$(echo "ibase=16; $soc" | bc) &&
mosquitto_pub -h localhost -t CANSee/RX/SOC -m $(echo "scale=2;($soc*0.02+3)" | bc -l)

odo=`echo "i7ec,222006" | nc 192.168.2.81 35000 -w 1 | cut -c 11-16 | tr '[:lower:]' '[:upper:]' ` &&
odo=$(echo "ibase=16; $odo" | bc) &&
mosquitto_pub -h localhost -t CANSee/RX/ODO -m $odo

rng=`echo "i7ec,223451" | nc 192.168.2.81 35000 -w 1 | cut -c 11-14 | tr '[:lower:]' '[:upper:]' ` &&
rng=$(echo "ibase=16; $rng" | bc) &&
mosquitto_pub -h localhost -t CANSee/RX/RNG -m $rng

fi
`
Here I have a question: what ist the reason BT an Wifi can not be on at the same time?
In my case wifi request take only place while charging.