fesch/CanZE #751: only dark theme in IOS version?

AndyIreland, 2022-08-01:
hello is there only a dark theme when using on Ipad IOS ? - I went into settings and could not find a setting to change to a light theme
Thanks.

ik2wxx, 2022-08-02:
Hi
What version of CanZE are you using?
Our iOS version has only light theme.

AndyIreland, 2022-08-02:
3.0.1.0
Thanks

[image]

ik2wxx, 2022-08-03:
I'm sorry that's not the official CanZE app so we can't help.

closed 2022-08-03 by ik2wxx

AndyIreland, 2022-08-03:
> I'm sorry that's not the official CanZE app so we can't help.

Oh, I am surprised at that, I downloaded it from the apple App icon on the ipad I thought it was the one.

edddeduck, 2022-08-03:
@AndyIreland 

There is a version of the code built by Frédéric Richard on the AppStore. This is using the same source code as a base from my understanding but is a separate branch and not directly linked to the developers working on CanZe in this project. 

There is the TestFlight only version from this GitHub which will say the developer is Roberto Sonzogni.

You can subscribe to the TestFlight version here: https://canze.fisch.lu/install-the-dongle/ This will allow you to install the "official" one.

AndyIreland, 2022-08-03:
ah right I see, thank you for that