fesch/CanZE #752: MQTT

PvdGulik, 2022-08-05:
Would be nice if all sort of data could be send by MQTT to domoticz, home assistant etc.

yoh-there, 2022-08-24:
I don't think there is much appetite to develop this, main reason being that CanZE only queries what's required by the UI. In other words, what would be spit out would be entirely dependent on what the end user is doing.

For what you request the Renault API would be more appropriate, though a lot slower and for the older model cars more limited and less reliable.

sanchosk, 2023-06-25:
Actually, it would be great if, let's say, canSee dongle could automatically connect to specific wifi (e.g. at home) and report data to MQTT for battery level, range, current charging power, etc.
The problem with Renault API is that for lots of Zoes the subscription is no longer even available. So while the car works perfectly and your app can (mostly) read the data while in car, I have no way how to connect it to my home.
Currently reading through your source code, trying to find out how to send command to ELM327 dongle to get the battery SoC and similar data and researching ELMduino on how to implement custom PIDs.
The idea would be to have one ESP32 in the garage, when the car arrives and the BT dongle is visible, it will try to connect and start to report the data to MQTT, where it can be consumed by HA or NodeRed, etc.
I think this would be appreciated by a lot of people.

yoh-there, 2023-06-25:
The CanSee dongle can do that. Not through mqtt but through http.

sanchosk, 2023-06-25:
Can you post some short description on how canSee does it?

yoh-there, 2023-06-25:
The CanSee dongle uses an esp32 for Bluetooth and Wifi. It's fairly low
level (you need to know the canbus IDs), but for that reason also flexible;
the dongle doesn't know anything functional. But it can connect to wifi,
provides http access (than canze van make use of btw). A bit of programming
in node-red or home assistant could do the trick.

Added bonus: canze will become at 4-10 times faster.

From memory (I am on my phone now): https://gitlab.com/jeroenmeijer/CanSee

sanchosk, 2023-06-25:
I am reading the source code and this CanSee should be pretty simple to mod to also publish to MQTT. My only problem is that I do not know the can IDs for battery SoC, range, charging power, etc. But still in process of reading the source code of the CanZE, so I might find it out later.

yoh-there, 2023-06-25:
I deliberately did not want to put the intelligence of what to send to the
car how to interpret the resulting data in the dongle. IMHO that is an
application responsibility, aka canze or the home automation.

Of course it can be done!

sanchosk, 2023-06-25:
Yep, that's exactly what I want - replace HTTP for MQTT and then "teach" the automation the codes to send/receive.
I found so far the response code for the CAN for Available discharge energy (427.49) in _Fields.csv file, but it does not include the PIDs I need to ask for - it seems I need to ask for some other PID to talk to BMS and that will then spit a lot of data, some of which will be this 427.49 value for Available discharge.
This seems quite promising, already ordering the 2 CAN bus controllers, have plenty of ESP32s around and also the DC2DC converter, so this will be fun :)
Now to find PIDs that the automation should ask and I am golden.

sanchosk, 2023-06-25:
Thanks for kicking me in the right direction. As I need to wait for the HW for CanSee to be built, I'll try to do something else.
I now can understand more or less the codes sent by AT commands in CanZE.
I can do the frame set header, the filter, etc. Then I cat set the request, for example to 0322502c, where the 22502c seems to be a code for PID 427.49 - from the ZOE_Ph2/_Fields.csv.
Then I'll get a response starting with 0562320cxxxxAAAA, in my example 0562320C0DC4AAAA.
This can then be separated from the response code, the real data will then be 0DC4.
According the same file (ZOE_Ph2/_Fields.csv), I need to multiply the value using coeficient 0.005 to get the kWh available.
And really, the 0DC4 translates to 3524 decimal, multiplied by 0.005 it makes 17.62kWh.

PERFECT!
I am SO close to have this ready to connect to Home Assistant, thanks again!

KarelSvo, 2023-06-26:
> how to send command to ELM327 dongle

############################################################
#[06/26/23 18:42:20.393] Init CAN
############################################################
AT WS
AT E1
AT S0
AT H0
AT L0
AT AL
AT CAF0
AT CFC1

############################################################
#connect to:  Addr:93
############################################################
AT CP 18
AT SH DADBF1
AT FC SH 18DADBF1
AT FC SD 30 00 00
AT FC SM 1
AT ST FF
AT AT 0
AT SP 7
AT AT 1
AT CRA 18DAF1DB
02 1003
03 229002

> custom PIDs
name,ShortName,ModeAndPID,Equation,Min Value,Max Value,Units,Header
"Battery USOC(Wxx_usoc_avg)","DID_$9002","229002","(A*256+B)*0.01","0","0","%","18DADBF1"

[bms.pid.zip](https://github.com/fesch/CanZE/files/11871802/bms.pid.zip)

KarelSvo, 2023-06-26:
I don't know which vehicle you are sending to. The parameter belongs to Zoe Phase 2 or other vehicles with RBMS Ecu.

sanchosk, 2023-06-26:
Sending to Zoe Phase 2 - build year 2014.
It seems OK if the CAN bus is offline (when the car is off), it does report CAN ERROR and NO DATA, when the car is on, it reports NO DATA and NO DATA.
Perhaps I am doing something wrong...

KarelSvo, 2023-06-26:
Phase 2 is from September 2019.

sanchosk, 2023-06-26:
You can clearly see I am totally aware of all the details :D 
Looking at the CanZE, I have it set to ZOE Q210 and it works (mostly, sometimes it fails to reset the ELM chip correctly).
Any chance you would know the correct sequence for my model of Zoe?

KarelSvo, 2023-06-26:
############################################################
#[06/26/23 19:38:27.389] Init CAN
############################################################
AT WS
AT E1
AT S0
AT H0
AT L0
AT AL
AT CAF0
AT CFC1
############################################################
#connect to:  Addr:93
############################################################
AT SH 79B
AT FC SH 79B
AT FC SD 30 00 00
AT FC SM 1
AT ST FF
AT AT 0
AT SP 6
AT AT 1
AT CRA 7BB
02 10C0
03 221405

_Battery USOC","221405","A/2.0","0","0","%","79B"

sanchosk, 2023-06-26:
This returned some data!!!

ELM serialBT: 0250C00000000000
ELM serialBT: 037F221100000000

KarelSvo, 2023-06-26:
ELM serialBT: 0250C00000000000 positive reply to 10C0

 ELM serialBT: 037F221100000000  negative answer "Service Not Supported"

02 10C0
03 221404   real battery SOC 
03 221406 SOH

sanchosk, 2023-06-26:
I will check the flow for the CanZE - how it communicates with the car when it tries to read the data.
It seems that for my car, the relevant line is 
,7ec,24,39,.02,0,2,%,222002,622002,ff,State Of Charge (SOC) HV battery
This field is something I can see both in debug and field log from CanZE.
Just need to find out how to set the filters for it.
I am SO CLOSE!!! Thanks a lot, Karel!

KarelSvo, 2023-06-26:
############################################################
#connect to:  Addr:95
############################################################
AT SH 7E4
AT FC SH 7E4
AT FC SD 30 00 00
AT FC SM 1
AT ST FF
AT AT 0
AT SP 6
AT AT 1
AT CRA 7EC
02 10C0
03 222002

sanchosk, 2023-06-26:
You are GENIUS!
056220020FF8AAAA => exactly what I've been looking for - 81.76% SoC!
Do you have some coffee/paypal or similar account, do that I can say "thank you" in credits?

sanchosk, 2023-06-28:
@KarelSvo Unfortunately, the code worked for 1 day and now all the commands receive OK, but the last one receives STOPPED>.
Any chance you have some idea what am I doing wrong?

sanchosk, 2023-06-28:
Fixed - removing the 02 10C0 step and only doing the 03 222002 returns the data no problem (as long as the car does not go to sleep :D).

closed 2023-10-02 by yoh-there

sanchosk, 2023-12-14:
@KarelSvo Sorry to bother you again (I really owe you a coffee already, or better some good Czech beer).
I am trying now to grab odometer status and report it to MQTT. All other reports (available kWh, available range) already work thanks to you.
However, for the odometer, I end up with timeout (no data received).
I tried to duplicate the items I have in log from communicating using CanZE with my Zoe, but I am failing.
Last time, you were the one able to find the "missing piece" in the AT commands. Perhaps you can do your magic one more time?

```
>  ate0
<  OK>

>  ats0
<  OK>

>  ath0
<  OK>

>  atl0
<  OK>

>  atal
<  OK>

>  atcaf0
<  OK>

>  atfcsh77b
<  OK>

>  atfcsd300000
<  OK>

>  atfcsm1
<  OK>

>  atsp6
<  OK>

>  atsp6
<  OK>

>  atsh79c
<  OK>

>  atcra7bc
<  OK>

>  03224b9b
<  NO DATA>
```

sanchosk, 2023-12-14:
Thanks, it took me a while to get the fact that it's the last 3 of 4 numbers that matter :)
But it works - now it's 2 beers/coffees.
`032220060662200601A497AA` - from this the 01A497 is the odometer :+1: 
Thanks again!