fesch/CanZE #753: Zoe 50: charge history not available + failure to get firmware versions

jouvin, 2022-08-23:
Hi,

I recently started to use CanZE with a KONNWEI 902 device to monitor my Zoe 50 (R135). Thanks a lot for this great application! It already help me to troubleshoot charging issues on domestic plugs and better understand the state of the battery or to get a more precise idea about several figures.

I noticed that the charge history remains empty (despite the total number of full and partial charges are available) and I see some "debug.. fail" message when opening this page. The same occurs if I try to look at the firmware versions: it works for some processors but fail for others and after this particular failure, I need to kill CanZE to reestablish the bluetooth connection which is lost after the error. Is it something known?

Thanks in advance for your help.

Michel

yoh-there, 2022-08-24:
Can you please indicate if you are using Android or iOS, and if iOS, the version of the team?

Also, and pardon me for asking the stupid questions, can you confirm the car type in the app is set to ZE50?

jouvin, 2022-08-24:
Hi,

Sorry for forgetting to mention that I was using Android (Android 12, Pixel 6 phone). Yes I have the car type set to ZE50 (in fact ZE40/ZE50).

Michel

yoh-there, 2022-08-24:
Thanks for clarifying.

Calling on the ZE50 specialists for a quick check - comparison please? @Leobaillard @FulminatingGhost

FulminatingGhost, 2022-08-24:
Hello,

I believe the charge history is only available up from a certain version of the EVC ECU. Please take a look what version of the EVC ECU your ZOE reports. You can check the version of the EVC from within CanZE. [Here](https://docs.google.com/spreadsheets/d/16RcvSsQHwWkZZ5pPrKsRNVgfjo_dUlNURQzFWRIitcA) is a spreadsheet of users tracking firmware versions inside their ZOEs which might also be interesting to you.

jouvin, 2022-08-25:
Thanks for your answers. Unfortunately I cannot access the EVC ECU firmware version. It is one of the ECU where CanZE reports an error when trying to access the ECU details. I tried the button "write everything in a CSV", it took some time to complete with a few debug errors but I don't know where to find the CSV if it has really been created. May be everything points to old firmware versions. If it is the case, how I can get them updated? My Zoe is 2 1/2 year old and it subscribed the annual maintenance. I'd expected that the firmware upgrade is part of what should be done by Renault... Any experience?

Michel

jouvin, 2022-08-25:
Thanks for the detailed answer. I'll try to ask for a firmware update at the next maintenance 8n a few months. I'll look at the issues you mentioned to argue with Renault about the update!

I found the CSV at the mentioned location and in fact, as I was running CanZE Plus, the location is org.bobby.canze. Both can only retrieve the UCH fw version, the 9thers generate an erro4 when sending atsh or atsp7 command. It may be that I am not running the very last build of CanZE (1.6.0 from Feb 2021) as CanZE Plus gives more details (1.63.50). In fact I had difficulties to find CanZE on Google Play and don't know how to get the last version.

Michel

jouvin, 2022-09-25:
@FulminatingGhost I managed to get an update of several firmware from Renault, thanks to all the information you pointed me to. It definitely improved the situation for CanZE, in particular the charge history is now available (filled in at each charge). The firmware information is also a little bit better: I can now get the information about several of them when clicking "Write all to CSV" (not sure about the exact text in English as I'm running the French version but it remains incomplete as show in the [attached CSV](https://github.com/fesch/CanZE/files/9640981/Firmwares-2022-09-23-17-53-42.xlsx). I noticed that after a failure to get one of the firmware (typically `error sending atsp6 command`), all the further attempts to get a firmware version, whether individually or usng the CSV button, will fail. In fact nothing is working in CanZE: it is necessary to reinitialize the bluetooth connexion to get it working again. I tried to write a debug file but unfortunately I have difficulties to export it (I am running Android 13 where the data directory is now really protected. I managed to access it but not to export the debug file somewhere else...).

Because of this problem I cannot really check what Renault installed and if it is matching what I was expecting. In fact I suspect that some updates are missing because I still don't see the Energy Information menu in Easylink.

Another problem I identified is that in the Brakes view, all the braking torque is reported as regenerative, whatever the torque value. I found this suspect but I guess it is coming from something improperly reported by the Zoe rather than from a flaw in CanZE.

Let me know if there is anything useful that I can do (apart from contacting Renault!) to fix the issues.

Michel

FulminatingGhost, 2022-09-25:
Congratulations on getting the updates!
I'm not very active with development of CanZE and I've only ever used the CanSee adapter with CanZE and no ELM327 so I don't have a lot of insight into the AT commands. Maybe someone else might be able to help here. If you feel up to it I would definitely recommend building your own CanSee adapter. It makes things a lot faster and more reliable.

After the EVC ECU firmware update the Radio will need a "Radio reset" in order to show the Energy info screen. This is an extra step required to be done by the garage. Maybe there's an other ways of doing that yourself but I have not done a lot on the car regarding this in quite some time so I'm a little bit out of the loop.

I'm afraid the brakes view does not show correct values for the PH2 Zoe and this was due to the PH2 Zoe not providing the required information but I might remember that incorrectly. Maybe we should disable or modify this view for the PH2? @Notice-Jeroen-Meijer

jouvin, 2022-09-25:
Thanks for your quick answer. Getting the update was surprisingly easy, after reading all the bad experiences on the forums. I wrote to the garage an email summarizing the most important ones and during the intervention somebody called me saying they had identified a lot of firmware requiring some update and that it will be done for free! Except one regarding the radio and display (but the price was symbolic! not even sure why they charge it!).

I have the feeling this last one was not completely done by the garage as there was still an Easylink update in progress when I got back the car. Thanks for the information on the "radio reset". I'll check with the garage!

jouvin, 2023-04-07:
Just wanted to give an update on my firmware upgrade adventure... Since 6 months I'm going roughly every month to the garage to get the car connexion fixed. Without any success. At last visit Renault said that that the TCU was dead (concretely means damaged by the update in September) and needed to be replaced. Renault accepted to take in charge a big part of the cost but not all as the car is more than 2 years. Fingers crossed that it will restore the normal services...

yoh-there, 2023-10-02:
Closing (stale)

closed 2023-10-02 by yoh-there