fesch/CanZE #754: zoe Z.E. 40

stefanozoe, 2022-08-26:
Non riesco a collegare il mio Vgate iCar ODB professional solution all'applicazione CanZe.
La connessione bluetooth funziona e la spia blu lampeggia.
La spia rossa rimane accesa fissa.
Mentre la spia verde lampeggia per 1 o 2 secondi e poi si spegne.
Qualcuno mi può aiutare?
Grazie

pieter2424, 2022-09-24:
I had the same problem. For me it worked through bluetooth with an iphone after using the vgate update app (there is a QR code on the vgate box i got). Now I dont even need wifi (bought the wrong vgate version) with my iphone

stefanozoe, 2022-09-25:
Thank you. I, on the other hand, solved it by changing my phone. The 
former was too old.

Il 24/09/2022 21:45, pieter2424 ha scritto:
> I had the same problem. For me it worked through bluetooth with an 
> iphone after using the vgate update app (there is a QR code on the 
> vgate box i got). Now I dont even need wifi (bought the wrong vgate 
> version) with my iphone

--

stefanozoe, 2022-09-25:
Thank you. I, on the other hand, solved it by changing my phone. The 
former was too old. stefano

Il 24/09/2022 21:45, pieter2424 ha scritto:
> I had the same problem. For me it worked through bluetooth with an 
> iphone after using the vgate update app (there is a QR code on the 
> vgate box i got). Now I dont even need wifi (bought the wrong vgate 
> version) with my iphone

--

closed 2023-10-02 by yoh-there