fesch/CanZE #755: Activity memory leak caused by anonymous threads

cuixiaoyiyi, 2022-08-31:
# Possible Memory Leak
An [anonymous inner class](https://stackoverflow.com/questions/17778077/would-anonymous-handler-or-runnable-create-a-memory-leak) will hold a reference to the this pointer of the outer class and will not be released until the thread ends.
It will hold the Activity and prevent its timely release. Please check the links below.

# Occurrences
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/widgets/WidgetView.java#L342
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/CanzeActivity.java#L258
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/CanzeActivity.java#L94
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/MainActivity.java#L524
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/MainActivity.java#L648
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/DtcActivity.java#L109
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/ElmTestActivity.java#L44
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/AllDataActivity.java#L111
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/FirmwareActivity.java#L123
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/TwingoTestActivity.java#L87
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/TwizyTestActivity.java#L83
https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/activities/TiresActivity.java#L282

# Possible Solution
If it is necessary, it can be changed to [static class + weak reference](https://stackoverflow.com/questions/22162642/will-handler-inside-a-method-leak-memory/27825703#27825703) to eliminate the reference to the activity, which may cause memory leaks.
Further discussion is welcome.

fesch, 2022-08-31:
Hi

None of the thing executed inside the newly created thread is time-consuming and so the latter will be finished far before the user closes the activity again. Thus I do not really understand how this could lead to some memory leak.

Could you perhaps give further details?

yoh-there, 2023-10-02:
Closed (stale)

closed 2023-10-02 by yoh-there